Screenly/Anthias #1007: Iframe not displaying consistently

xanbrennan, 2019-01-08:
I have a few pages set to load embedded Google Sheets through OSE on a Raspberry Pi 3. The iFrames inconsently load with about 1/3 of the time them just showing white. I’m at a loss as to how to make the pages load consistently.

One of the pages -
http://www.goldenbeeproperties.com/digitalscreen01/Sheet1.html

vpetersson, 2019-01-08:
@xanbrennan - Could you try switching to the experimental branch to see if the problem persists?

xanbrennan, 2019-01-08:
I'm running Sprint 8. I'm not sure where to find the experimental branch.

vpetersson, 2019-01-08:
Log into the device and run the upgrade script (`bash <(curl -sL https://www.screenly.io/install-ose.sh)`. Then select the experimental branch when prompted for what version you want to install.

xanbrennan, 2019-01-08:
I'm sorry, total NOOB. I can flash the SD card, but not sure how/comfortable with loging to the terminal from my Mac.

vpetersson, 2019-01-09:
Alright, in that case i think you will need to wait until that branch becomes stable.

ealmonte32, 2019-01-09:
@xanbrennan 

I'll try and help you, this shouldn't be that hard, even for noobs.

You did not mention if you had physical access to the rpi or if ssh was enabled on it..

To run the upgrade, you either do it by physically using the rpi or via ssh.

There a few ways to enable ssh (_so that you can run commands on the device remotely over the network_) on the rpi, here are two of them, whichever one sounds easier to you then go for it:

**Method 1:**
1. Get into terminal/console of the pi (not your mac), by using a keyboard connected to the rpi running screenly ose, and press ctrl+alt+f1 , then log into the pi with username pi and unless you changed the default password, should be raspberry.
2. Type: sudo raspi-config , then press enter
3. Select Interfacing Options.
4. Navigate to and select SSH.
5. Choose Yes.
6. Select Ok.
7. Choose Finish.

**Method 2:**
1. Mount the SD card of the rpi to a computer
2. Go into the /boot/ partition
3. Create an empty file and name it ssh or ssh.txt (case sensitive)
4. That's it, that system will now have SSH enabled when booted

**To perform the upgrade vpetersson mentioned, when on the terminal, type this:**
sudo ./screenly/bin/run_upgrade.sh

then follow on screen instructions..

_just for reference_
_full directory /home/pi/screenly/bin_

_if you press tab while starting to type a word it auto completes it for you if it exists on the directory_

_this info is also on the screenly wiki page: https://github.com/Screenly/screenly-ose/wiki_

xanbrennan, 2019-01-10:
Thanks, but this became to much work for what I was looking for so I switched to info-beamer. 

I'll come back to Screenly later when I have time to play with it.

closed 2019-01-10 by xanbrennan