Screenly/Anthias #1013: Screenly producing 2 ip addresses

zephyr96, 2019-01-08:
I have about 20 R-PI 2's running screenly in our plant using an Ethernet connection. Recently we have started having issues with running out of DHCP addresses. We have about 100 DHCP IP's on our network that we mostly use for WiFi. After installing screenly, I set each one up with a static IP address. This week when checking the network I noticed multiple pi's on our DHCP block. I double checked half of them and all came back the same. My PI's running screenly have 2 IP addresses. One static that I set and 1 DHCP. I have never seen this before and have been trying to figure out the how and can not. Any help would be great.

* Raspberry Pi Hardware Version: raspberry pi 2 b+ 
* Screenly OSE Version: v0.9.1

zephyr96, 2019-01-08:
pi@10.0.0.90's password:

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Tue Jan  8 13:28:44 2019 from jbmlt-266.7coin.com
pi@ds-hr:~ $ ifconfig
eth0      Link encap:Ethernet  HWaddr b8:27:eb:a2:09:26
          inet addr:10.0.2.85  Bcast:10.0.3.255  Mask:255.255.252.0
          inet6 addr: fe80::ba27:ebff:fea2:926/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:126928750 errors:0 dropped:479773 overruns:0 frame:0
          TX packets:12462570 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:1354680235 (1.2 GiB)  TX bytes:1248367195 (1.1 GiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:3185 errors:0 dropped:0 overruns:0 frame:0
          TX packets:3185 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:4159925 (3.9 MiB)  TX bytes:4159925 (3.9 MiB)

wlan0     Link encap:Ethernet  HWaddr b8:27:eb:f7:5c:73
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:9 errors:0 dropped:9 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:1307 (1.2 KiB)  TX bytes:0 (0.0 B)

vpetersson, 2019-01-08:
From the output of that, this device does not have more than one IP address (10.0.2.85).

zephyr96, 2019-01-08:
If you look at the top, "pi@10.0.0.90's " I'm connected to it via IP address 10.0.0.90. I can connect to it using either IP address. Same for the web portal I get to the same pi using either address too. To make things even more complicated. arp comes back for 10.0.0.90, but not 10.0.2.85. 10.0.0.90 has the same MAC as shown above.

JonnyTech, 2019-01-08:
Is `10.0.0.90` its hostname?

zephyr96, 2019-01-08:
> Is `10.0.0.90` its hostname?

No, hostname is ds-hr

ealmonte32, 2019-01-08:
@zephyr96 can you post the output of nmcli d

zephyr96, 2019-01-08:
> > Usually terminal prompt is `user@hostname` but your output shows `pi@10.0.0.90` and `pi@ds-hr` - from the same unit?
> > JohnnTech, when using putty I can use either the ip or the hostname. which ever one I do use, that one shows up. so if I used 10.0.2.85 it would show up as 'pi@10.0.2.85' yet it is connecting to the same pi.
DEVICE  TYPE      STATE         CONNECTION

wlan0   wifi      disconnected  --

eth0    ethernet  unmanaged     --

lo      loopback  unmanaged     --

ealmonte32, 2019-01-08:
> wlan0 wifi disconnected --
>
> eth0 ethernet unmanaged --
>
> lo loopback unmanaged --

as far as i know it is impossible to have one interface (eth0) have two different IP addresses with the same MAC address.. now, that being said, i have read about some virtual configurations and things some people have tried to do with one interface but multiple IPs using a virtual adapter method, so assuming this is def not something you are doing, lets proceed..

can you show us if possible the following outputs: (you can redact the private parts)

sudo cat /etc/network/interfaces

and

(assuming you are using network manager)
sudo cat /etc/NetworkManager/system-connections/*whatever is listed here*

zephyr96, 2019-01-09:
@ealmonte32 
# Generated by Screenly Network Manager
auto lo
  iface lo inet loopback

auto eth0
  iface eth0 inet static
  address 10.0.2.85
  netmask 255.255.252.0
  gateway 10.0.0.1

When If would run sudo cat /etc/NetworkManager/system-connections/ It would come back as no such file or directory. 
@vpetersson : pi@ds-hr:/ $ traceroute 10.0.0.90
traceroute to 10.0.0.90 (10.0.0.90), 30 hops max, 60 byte packets
 1  ds-hr.7coin.com (10.0.0.90)  0.109 ms  0.037 ms  0.035 ms
pi@ds-hr:/ $ traceroute 10.0.2.85
traceroute to 10.0.2.85 (10.0.2.85), 30 hops max, 60 byte packets
 1  10.0.2.85 (10.0.2.85)  0.138 ms  0.038 ms  0.032 ms

vpetersson, 2019-01-09:
Honestly, this beats me. There is no indication on the device that it is listening to that IP. Hence I would guess that it's something in the network. Perhaps there is some port forwarding setup. Could you share some insight on how the network is configured (ideally with hardware vendors/models etc).

robomir, 2019-01-09:
You need to use "ip addr show" or "ip a s" for short. There's no problem to have multiple ips on the same interface. Also can you post your /etc/network/interfaces file? And maybe run "ps axf | grep dhclient"

ealmonte32, 2019-01-09:
> @ealmonte32:
> 
> > as far as i know it is impossible to have one interface (eth0) have two different IP addresses with the same MAC address
> 
> No, you can actually with IP aliasing (just do `ifconfig eth0:0 192.168.a.b`). However, I don't think that is the case here.
> 

Thanks for that IP aliasing info viktor, I thought only virtual adapters could do that.. 

@zephyr96 
># Generated by Screenly Network Manager
>auto lo
>  iface lo inet loopback
>
>auto eth0
>  iface eth0 inet static
 > address 10.0.2.85
 > netmask 255.255.252.0
>  gateway 10.0.0.1
>When If would run sudo cat /etc/NetworkManager/system-connections/ It would come back as no >such file or directory.

To me this proves that there is no virtual or IP aliasing going on here..
It clearly shows that the assigned IP of the host "ds-hr" is 10.0.2.85.
An empty systems-connections directory just means you have not created a connection using network manager, which is ok.

@robomir he already provided the contents of the interfaces file - see above.

and yeah i also think it is something on the dhcp end.. or something on the network config and DNS.. 

@zephyr96 can you show us the nslookup of both of those IP addresses?

robomir, 2019-01-09:
Thank you @ealmonte32, somehow I've missed the post. 
I'm convinced it's the pi's asking for a dhcp assignment since there's no way a server will just configure the ip there. In order to find that out we just need to check for running dhclient with "ps afx | grep dhclient" also "ip a s" must show both IPs on eth0

zephyr96, 2019-01-09:
@vpetersson @ealmonte32 @robomir 

Sorry for the delay, 
pi@ds-hr:~ $ ps afx | grep dhclient
14935 pts/1    S+     0:00              \_ grep --color=auto dhclient

 ip a s
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether b8:27:eb:a2:09:26 brd ff:ff:ff:ff:ff:ff
    inet 10.0.2.85/22 brd 10.0.3.255 scope global eth0
       valid_lft forever preferred_lft forever
    inet 10.0.0.90/22 brd 10.0.3.255 scope global secondary eth0
       valid_lft forever preferred_lft forever
    inet6 fe80::ba27:ebff:fea2:926/64 scope link
       valid_lft forever preferred_lft forever
3: wlan0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN group default qlen 1000
    link/ether b8:27:eb:f7:5c:73 brd ff:ff:ff:ff:ff:ff

ealmonte32, 2019-01-09:
and there it is... good job @robomir 
 ip a s command indeed shows all IPs of the interface

ealmonte32, 2019-01-09:
@vpetersson 
I have to say, I just tested this myself and it is true, screenly ose is automatically adding a secondary address..

ealmonte32, 2019-01-09:
@zephyr96 

I just finished researching this issue which others had back in 2015..
it seems you need to remove the static ip info out of /etc/network/interfaces and into the /etc/dhcpcd.conf file instead.. let me know if that works for you..

robomir, 2019-01-10:
So this is a os config issue after all. Why is the network configuration file changed from the default debian way is beyond me... I just hate it when os devs are doing such minor but utterly confusing changes.

vpetersson, 2019-01-10:
Right, yeah this was a strange decision upstream I must admin. Alright, so we are on top of this? 

Thanks @ealmonte32 and @robomir for the help.

zephyr96, 2019-01-10:
@vpetersson @robomir @ealmonte32 Thanks, got it deleted, rebooted used static ip to connect and ip a s is only showing static ip now. Thanks for the article @ealmonte32  Already applied the fix to two of them so far with no issue. Thanks again for the help!!

closed 2019-01-10 by zephyr96

ealmonte32, 2019-01-10:
> Right, yeah this was a strange decision upstream I must admin. Alright, so we are on top of this?
> 
> Thanks @ealmonte32 and @robomir for the help.

@vpetersson are you thinking of making changes to the build image and edit the default /etc/network/interfaces file to warn the users of screenly in big capital letters that they should not set any static IP info there?.. although the file by default comes with a comment already, I must say that it is not very strong in warning users to really not use that file for that purpose..

vpetersson, 2019-01-10:
Yes, we might need to make a few improvements to this. However, as of the last year or so, the "Screenly Network Wizard" (as shown in the config above) is not used for OSE anymore. Hence, this somewhat solves the issue as the network configuration falls outside the scope of Screenly. However, additional documentation is of course always good.