Screenly/Anthias #1014: Javascript Date() keeps returning GMT+0

Sambiki, 2019-01-09:
Hi,

We are using our own page as signage with experimental built of Screenly. Despite having set up the local time correctly, the local time (javascript) on the page keep returning GMT+0. This is not the case when opened from another device. Anyone having the same issue? One more thing, we tried with stable built, and the time was correct, but some of CSS just didn't work. Here is the script:
`var d = new Date()`
`document.getElementById("time").innerHTML = d;`

rusko124, 2019-01-09:
Hi @Sambiki !
I can offer you to set the necessary timezone in `raspi-config`
This will be fixed with the next image release.
You can view progress here https://github.com/Screenly/pi-gen/issues/11

vpetersson, 2019-01-09:
(@rusko124 the "locale" has nothing to do with the timezone.)

vpetersson, 2019-01-09:
>  My mistake, I thought that Screenly image build will set the default value for timezone too.

No, you're right about that part. It is just another setting.

Sambiki, 2019-01-10:
Thanks. I believe the local settings is correct. So, i can only wait for the next image release then?

rusko124, 2019-01-10:
@Sambiki 
You can still change the timezone(not locale) in raspi-config. This should help you with your problem.
I reproduced it, and in my case it helped me.

ealmonte32, 2019-01-10:
@Sambiki same here, had the same issue and since locale deals with language framework, you def have to change the timezone settings:
sudo raspi-config
option 4 - localisation options
option 12 - change timezone

when you finish i recommend a reboot.. if that doesn't help let us know..

bhagemann, 2019-01-14:
So having similar? ongoing time issues here (reported last summer), 2 different units with symptoms as follows:

1 Raspi3 with sprint 7
1 Raspi3 with sprint 8 then upgrades with experimental branch

I'm in US/Eastern timezone.

Initially the units show as being in UTC time which displays 4 hrs ahead of where we are using the date command in the shell.

Screenly displays the following
Built-in Time Widget - 4 hrs ahead
Built-in Weather Widget - CORRECT time

my local Solar panel dashboard [https://monitoringpublic.solaredge.com/solaredge-web/p/kiosk?guid=4df18913-3506-4aea-9208-e576bb78003f]
4 hrs ahead on Screenly but correct in my laptop browser

boardclock demo webpage [https://boardclock.com/app/clock/626fcd71cd166954225cd2b703bd22c8/]
4 hrs ahead on Screenly but correct in my laptop browser

I've used the raspi-config utility to change the timezone to Eastern time.  Now the date command in the shell shows the correct time, BUT no change on any of these websites that display the time in Screenly. 

Been experiencing this since last spring, I THINK an earlier release reflected the proper time, but not sure.

Any ideas appreciated, I can't find a way to get this to show time for us in our Lobby here and when the Solar Kiosk shows up, it always has the incorrect local time which folks keep pointing out to me.

vpetersson, 2019-01-14:
@bhagemann Interesting observation. I don't see how that could be. @rusko124 can you ake a look?

ealmonte32, 2019-01-16:
@bhagemann 
Can you pls post the output of command timedatectl ?
Thx

bhagemann, 2019-01-16:
pi@raspberrypi:~ $ timedatectl
      Local time: Wed 2019-01-16 12:55:00 EST
  Universal time: Wed 2019-01-16 17:55:00 UTC
        RTC time: n/a
       Time zone: America/New_York (EST, -0500)
 Network time on: yes
NTP synchronized: yes
 RTC in local TZ: no
pi@raspberrypi:~ $

bhagemann, 2019-01-16:
with that, the screenly clock app shows 17:55, the weather app shows 12:55, my solar array 17:55, the dashclock someone posted the link to 17:55.  On my laptop, web browser shows 12:55 on all I try.

ealmonte32, 2019-01-16:
@bhagemann 
Thanks for info.

I have replicated the same issue with https://clock.srly.io/ but my https://www.time.gov and the https://weather.srly.io both show the correct 12-hour eastern time..

The clock app may go to 24-hour by default, since I get it as 24-hour on my computer as well as the rpi, all you have to do is make your asset this:
https://clock.srly.io/?24h=0

That will show the clock in 12 hour format with correct timezone.
You could even add the ?24h=0 to the end of weather app URL as well and it will force it 12-hour..

Can you give me the URLs of all the websites you try and get UTC time instead of EST time on your raspberry pi? I want to try them on my screenly pi and see.

bhagemann, 2019-01-16:
OK... glad to know about the 12/24 hr toggle.  But first to get the right time ! ;-)

This came up as an issue first with our Solar Array Dashboard which has a time of last update on it and shows the UTC time.  The url is:

https://monitoringpublic.solaredge.com/solaredge-web/p/kiosk?guid=4df18913-3506-4aea-9208-e576bb78003f

That and the two clock urls should be enough to figure out how to fix I hope.  Clearly something about the way these 3 sites work is different than the weather app url and displaying the UTC instead of the correct local time.  Thanks for looking into it!

rusko124, 2019-01-28:
@ealmonte32 @bhagemann More options for widgets you can see [here](https://app-store.srly.io/) 
@vpetersson I confirm what @bhagemann said. The clock widget takes timezone of system. The weather widget takes timezone by ip location.

ealmonte32, 2019-01-28:
@bhagemann 
I am running the experimental version but I cannot load the solar array page.

What does your screenly pi display on the right frame if you set an asset to this page and test it:
https://www.w3schools.com/jsref/tryit.asp?filename=tryjsref_date_obj

Does it return something like this:
Mon Jan 28 2019 16:26:09 GMT-0500 (EST) ?

bhagemann, 2019-02-04:
so yeah the solar array dashboard now shows a blank white screen in screenly both production version and experimental - thats a new issue, it used to work (but with wrong time) and continues to work in any web-browser on my mac or a pc.  something might have changed in their website code that broke it in screenly?  thats gonna kill it for my lobby display.

But back to the clock:

that URL you gave returns
Mon Feb 04 2019 15:45:23 GMT-0 (UTC)

and I still get the correct time with the Screenly Weather app and wrong (+5) time with the screenly time app.  Other websites with time in them give either one or the other, but can't find a commonality that i can see from this end

ealmonte32, 2019-02-08:
> so yeah the solar array dashboard now shows a blank white screen in screenly both production version and experimental - thats a new issue, it used to work (but with wrong time) and continues to work [...]

As far as I know, this most likely has to do with the browser+version.
Can you do me a favor and paste the outcome of the following command:

sudo update-alternatives --config x-www-browser <i>(just press enter after but when it asks to make a selection just press enter again leaving the current selection as-is)</i>

Then, to see what version of uzbl browser you are using, type this command:

sudo apt-get install uzbl <i>(it should say you already have it installed but i would like to see the exact version info)</i>

-----

> But back to the clock:
> 
> that URL you gave returns
> Mon Feb 04 2019 15:45:23 GMT-0 (UTC)
> which is 5 hrs off the correct time (-5/EST)
> 
> and I still get the correct time with the Screenly Weather app and wrong (+5) time with the screenly time app.

This proves what rusko said, the screenly weather app is getting the time from ip location, not from your systems date/time so that is why it displays correctly, and the screenly time app gets it wrong because it gets it from your system, but what doesnt make sense is that your timedatectl shows correct info..
can you paste the outcome of this:

systemctl status systemd-timesyncd.service

-----

> Other websites with time in them give either one or the other, but can't find a commonality that i can see from this end

The URL I gave you is a simple javascript that calls the system's locale and time to display it which the browser just reads from system.. that should be probably the only URL you should test from now on because once that tells you the correct time in EST then your system should be set..

ealmonte32, 2019-02-25:
@bhagemann - i  think you should give this package a try and see if it helps with time zone issue:

`sudo apt-get install htpdate`

If this doesnt work, I cant help unless you reply back with results/output from the other commands I posted on previous post in order to try isolating the issue.. also, since this is probably not a screenly ose issue and more of a configuration setting on the Pi, this should really be moved to the screenly ose forums that just opened:
`https://forums.screenly.io`

bcherup, 2019-03-22:
This issue has been fixed, it was a configuration with the docker settings. 
Try the following : 
   git reset --hard
   git pull
   sudo shutdown -r now

vpetersson, 2019-03-23:
Sounds like the fix worked. Closing out the issue.

closed 2019-03-23 by vpetersson