Screenly/Anthias #1016: Experimental browser, web components

dauledk, 2019-01-10:
#### Overview of the Issue
I can't load a webpage with [web components](https://caniuse.com/#search=web%20components), I am running the experimental version of Screenly.

#### Reproduction Steps

Steps to reproduce this issue, eg:
* Update to experimental
* Press Add Asset
* Enter the Asset URL ...
* Press "On"
* Wait for the asset to show up on the screen

### Environment

* Raspberry Pi Hardware Version: 3
* Screenly OSE Version: (I am not sure where I can find this info)

ealmonte32, 2019-01-10:
@dauledk 
I just tested this in non experimental and it works fine.

dauledk, 2019-01-10:
Hi @ealmonte32 Sorry for the confusing link. This link just show support for Shadow DOM v1 on different browsers. Can you maybe check [this link](https://jernbanecafeen.dk/arkiv/)? 
 
This part of the page is a web component.

[image]

ealmonte32, 2019-01-10:
Forgot to mention, the experimental branch does not come with uzbl but instead with alternative browser, as mentioned here:
https://github.com/Screenly/screenly-ose/issues/527

and here:
https://github.com/Screenly/screenly-ose/issues/824

Therefore, if the experimental branch browser does not render the component/elements on that page, i'm afraid at this moment that's the only option to at least have a newer browser than the one that comes with the regular version..

I will try some things and if i can come up with workaround i'll mention it here, if not, sorry =[

closed 2019-03-18 by stale[bot]