Screenly/Anthias #1019: Jitsi (Video conference) Webpage not loading

fa-grosch, 2019-01-17:
#### Overview of the Issue

When adding a [Jitsi meeting video conference link](https://meet.jit.si) into screenly as a new asset, it only displays a grey page without joining the video conference.

#### Reproduction Steps

Steps to reproduce this issue, eg:

* Press Add Asset
* Enter the Asset URL https://meet.jit.si/SeriousBearsHugLovingly
* Press "On"
* Wait for the asset to show up on the screen
* Only a grey page is displayed

### Environment

* Raspberry Pi Hardware Version:
3 Model B (a22082, Q1 2016), CPU: 1.2 GHZ, RAM: 1 GB

* Screenly OSE Version: 62bfe2ff477122ee97bf1b1017e94abffad52013
```
cd ~/screenly && git rev-parse HEAD
```

vpetersson, 2019-01-17:
@fa-grosch I doubt that will work. The Raspberry Pi is nowhere near powerful enough to handle something like a video conference stream.

fa-grosch, 2019-01-18:
@vpetersson thanks for your answer, would it work with a more powerful hardware + screenlyOS?

vpetersson, 2019-01-18:
Currently, we only support the Raspberry Pi officially. However, in theory, any of the supported hardware that [Balena](https://www.balena.io) could work.

closed 2019-01-18 by fa-grosch