Screenly/Anthias #1020: Extended character set file names

renowden, 2019-01-17:
Screenly OSE latest version (Sprint 8). Uploading assets from Windows 10. It objects to a file name containing an apostrophe (otherwise known as a right single quote).  This is whether by select file or drag and drop. It used to work on our previous version (before the drag and drop interface). Sample file name

Ricks’s picture.jpg

madsrh, 2019-01-18:
I can confirm this issue when using danish characters like æ, ø and å in the filename

`Server Error: 'ascii' codec can't encode character u'\xe6' in position 43: ordinal not in range(128)`

vpetersson, 2019-01-18:
Interesting find. I thought we had already solved this with proper UTF8 support. @rusko124 let's add this to the next sprint.

rusko124, 2019-04-26:
Solved and already in master branch.
@vpetersson I think we can close this.
@renowden @madsrh If this issue will not solved for you or you'll have any questions, please let me know.

closed 2019-04-27 by vpetersson