Screenly/Anthias #1022: Iframe

icarrasquedo, 2019-01-24:
I'm Having the next issue. I have a simple web page that usses the <iframe> tag, when runnig it from any browser I get the displayed image at it supose to be.  (Showing the inner page in a frame inside a webpage) nevetheless when I setup screenly OSE "NO" page inside the <iframe></iframe> tag is showed.
I have asked this to the screenly suppor people, even shared my URL for test but they told me that its working just fine on Screenly Pro.

the url is [https://86ee4b9d.ngrok.io/OMA/api/v1.0/fids/gps/localiza](https://86ee4b9d.ngrok.io/OMA/api/v1.0/fids/gps/localiza))

code for the page is 

```
<html><head></head><body> <iframe id="inlineFrameExample"
  | title="Inline Frame Example"
  | frameborder="0"
  | scrolling="no"
  | width="800"
  | height="600"
  | src="/static/html/map.html"></iframe></body></html>
```

so what could be wrong or is it that the OSE does not support the <iframe> tag?

vpetersson, 2019-01-28:
@icarrasquedo The browser in OSE is getting somewhat dated, and we are addressing this in the "Experimental version". You can try testing it by running the upgrade and select the experimental version. There is no explicit limit on iframes by any means to my knowledge however.

ealmonte32, 2019-03-28:
> I'm Having the next issue. I have a simple web page that usses the <iframe> tag, when runnig it from any browser I get the displayed image at it supose to be. (Showing the inner page in a frame [...]

Are you sure that iframe code works in other browsers? I've never seen separators | be used, only ; semicolons..

ealmonte32, 2019-03-28:
> I'm Having the next issue. I have a simple web page that usses the <iframe> tag, when runnig it from any browser I get the displayed image at it supose to be. (Showing the inner page in a frame [...]

@icarrasquedo 
Ignacio, creo que te falto un . antes de /static/html... porque para que el iframe busque en el mismo source de la pagina original, osea cuando le diste Add Assets, el URL que pusiste fue:
`https://86ee4b9d.ngrok.io/OMA/api/v1.0/fids/gps/localiza` ?
Si fue ese, entonces ./static/html/map.html tiene que estar disponible en ese server en el folder /gps/

ealmonte32, 2019-03-29:
@icarrasquedo 
Mira, este es el mismo test echo con un local server y usando screenly-ose, como puedes ver los iframe trabajan bien, algo esta mal en tu code and directory..

[image]

JonnyTech, 2019-03-29:
OT: @ealmonte32 can you please post in English (along with something else if you really need to) so that the majority of us can follow the conversation too?

ealmonte32, 2019-03-29:
> OT: @ealmonte32 can you please post in English (along with something else if you really need to) so that the majority of us can follow the conversation too?

Oh sure, I did not know others were following the conversation and were also having the same issue as the OP.
Since he is based in spanish speaking country and maybe using translator to post, I thought it would be easier to just post my reply help in spanish..

I have edited the posts to include the english version of my sentences..

vpetersson, 2019-04-01:
@ealmonte32 I agree with @JonnyTech let's stick to English in all comms to make sure others can follow along. A for effort tho :)

vpetersson, 2019-04-27:
Closing out issue due to lack of response.

closed 2019-04-27 by vpetersson