Screenly/Anthias #1024: fail to upload a new jpg file to cover the old one

unfirewood, 2019-01-29:
I have read the document at api/docs，  I can use “curl command” to modify the assets's attribute.   But I don't know how to upload a file by using "curl command".  I see some api such as "/api/v1/file_asset" ,but  I don't know how to use it.  I mean if I can use api to upload a new file (such as name: a.jpg) to cover the old file (name:a.jpg) ?

if I use the command:   curl -X GET "http://192.168.79.35/api/v1.1/assets" -H  "accept: application/json"   it success ,    and "curl -X POST "http://192.168.79.35/api/v1.1/assets" -H  "accept: application/json" -H  "content-type: application/json" -d "{  \"is_enabled\": 0,  ......."   it success.

but I use the command "curl -X POST "http://192.168.79.35/api/v1/file_asset" -H  "accept: application/json" -H  "content-type: application/json" -d " { \"File path\": \"aa.jpg\" }"  it failed and show "{"error": "'NoneType' object has no attribute 'filename'"}"

unfirewood, 2019-01-29:
I use a curl tool which can provide a method to send some "curl command" to screenly .  it run in windows dos

curl command1: 
C:\Users\unfirewood\Desktop\Curl>curl -X GET "http://192.168.79.35/api/v1.1/assets" -H  "accept: application/json"

then it return:
[{"asset_id": "2e9e39000508458bb740f25f5bfd90d2", "mimetype": "image", "name": "show1.jpg", "end_date": "2019-12-21T07:17:38.888000+00:00", "is_enabled": 1, "nocache": 0, "is_active": 1, "uri": "/home/pi/screenly_assets/2e9e39000508458bb740f25f5bfd90d2", "skip_asset_check": 0, "duration": "10", "play_order": 0, "start_date": "2019-01-21T07:17:38.888000+00:00", "is_processing": 0}]

command 2:
C:\Users\unfirewood\Desktop\Curl>curl -X POST "http://192.168.79.35/api/v1/file_asset" -H  "accept: application/json" "{  \"type\": \"file\", \"filename\": \"C:/Users/unfirewood/Desktop/IMAG0014.JPG\" }

it return:
{"error": "'NoneType' object has no attribute 'filename'"}
[1/2]:   "type": "file" --> <stdout>
--_curl_--  "type": "file"
curl: (3) URL using bad/illegal format or missing URL

[2/2]:  "filename": "C:/Users/unfirewood/Desktop/IMAG0014.JPG"  --> <stdout>
--_curl_-- "filename": "C:/Users/unfirewood/Desktop/IMAG0014.JPG"
curl: (3) URL using bad/illegal format or missing URL

maybe i took a wrong curl command format，  so could you show me the full curl command about /api/v1/file_asset   .   In my project,  I try to control screenly by windows software, it can update the contents itself.

rusko124, 2019-03-01:
Hi @unfirewood !

Image: `test.jpg`

### 1: Upload file
*IN:*
```
curl \
	-H "Content-Type: multipart/form-data"  \
	-X POST \
	-F "file_upload=@test.jpg" \
	http://x.x.x.x/api/v1/file_asset
```
*OUT:*
```
"/home/pi/screenly_assets/test.jpg.tmp"
```

### 2: Add to database
*IN:*
```
curl \
    -H "Content-Type: application/json" \
    -X POST \
    -d '{"duration": "10", "end_date": "2019-03-31T06:39:00.000Z", "is_active": 1, "is_enabled": 0, "is_processing": 0, "mimetype": "image", "name": "test.jpg", "nocache": 0, "play_order": 0, "skip_asset_check": "0", "start_date": "2019-03-01T06:39:00.000Z", "uri": "/home/pi/screenly_assets/test.jpg.tmp" }' \
    http://x.x.x.x/api/v1.2/assets
```
*OUT:*
```
{"asset_id": "9054ea3fdc9c4a73bc707ffebb0db021", "mimetype": "image", "name": "test.jpg-1", "end_date": "2019-03-31T06:39:00+00:00", "is_enabled": 0, "nocache": 0, "is_active": 0, "uri": "/home/pi/screenly_assets/9054ea3fdc9c4a73bc707ffebb0db021", "skip_asset_check": 0, "duration": "10", "play_order": 1, "start_date": "2019-03-01T06:39:00+00:00", "is_processing": 0}
```

unfirewood, 2019-03-07:
Thank you very much for your helping.
Now I can upload a jpg file to pi and the add an asset to database then It show,  it works by curl command step by step.

And then, I have a debt about that:  
1.   If there is a jpg such as "/home/pi/screenly_assets/test.jpg.tmp" store in pi, then I add an file test.jpg to pi, with the same file name,   the new file cover the old one, or rename as test2.jpg.tmp.
2.   For the asset,  it contain the jpg file content, or save a indicator direct to the jpg file

vpetersson, 2019-03-07:
@rusko124 are we really storing the filenames as they are uploaded? We should really assing a UUID to the filename on disk instead and return that. Moreover, when you are interacting with the asset (e.g. enabling it), that should be done with the UUID.

vpetersson, 2019-04-27:
Sounds like this issue has been resolved. Closing out. Feel free to comment if it is still an issue.

closed 2019-04-27 by vpetersson