Screenly/Anthias #1028: Suggested URL Format for Web Asset not working

farhangithub27, 2019-02-07:
#### Overview of the Issue

web asset with login parameters as suggest in #515 are not working. I have posted my detailed comment in #515  as well.

### Environment

* Raspberry Pi Hardware Version:pi3B
* Screenly OSE Version:image_2018-11-23-Screenly-OSE-lite

vpetersson, 2019-02-21:
@farhangithub27 I've added this to Sprint 9. We're a bit behind on this due to work being done on the commercial version of Screenly.

rusko124, 2019-04-24:
Hi @farhangithub27!
I reproduced all the variations of the urls you described above. 
In my case, all worked.

**Server Error: Could not retrieve file. Check the asset URL**
I get this error in the following cases:
- Invalid url
- `username:password@` not specified or invalid
In other cases, I have successfully added links and they are also successfully displayed in the viewer.

If you haven't solved this issue yet, try upgrading to the latest development version or you can provide a url that doesn't work in your case so I can test.

closed 2019-08-18 by stale[bot]