Screenly/Anthias #1030: Keyboard issue

hycsa, 2019-02-11:
#### Overview of the Issue

Hello, I am working on a school project with Digital Signage, I'm using Screenly OSE for this. I've installed Screenly on the Raspberry Pi, got on the dashboard and made a couple of assets to display on a monitor, this all went well. Now I have to show a sharepoint on the screen, which I made an asset for, but I have to log in first... The issue is I can't log in, I plugged my keyboard and mouse into the Pi and I am able to use the mouse but I cannot use my keyboard, I cant type anything. I can turn on capslock and such and even open up the terminal with CTRL+ALT+F1.

#### Reproduction Steps

Steps to reproduce this issue, eg:

* Press Add Asset
* Enter the Asset URL ...
* Press "On"
* Wait for the asset to show up on the screen
* Login screen to identify yourself
* Cannot type my credentials because I can't type on Screenly.

### Environment

* Raspberry Pi Hardware Version: Raspberry Pi 3 Model B+
* Screenly OSE Version: Latest release on: https://github.com/screenly/screenly-ose/releases (0.16)

Anyone who's had this problem and solved it or anyone who knows how to solve this? 
Thanks in advance!

vpetersson, 2019-02-12:
@hycsa This is a known limitations. For security reasons, HID devices are disabled. @rusko124 do you know that status of this?

hycsa, 2019-02-12:
> @hycsa This is a known limitations. For security reasons, HID devices are disabled. @rusko124 do you know that status of this?

Is there any workaround for this or any other possible option? The only thing holding me back right now is the login screen for the sharepoint, If I'm able to log-in to the sharepoint i'll be done with the assignment..

outtasync, 2019-02-12:
I am playing with Screenly too, and I am stuck on a web page asset login screen.   Just need to be able to enter a username and password and keep it logged in.

ealmonte32, 2019-02-20:
@hycsa - Can the sharepoint be "published" so that it gives you a public accessible URL instead of having to log in to show the sharepoint asset?

@outtasync - which page? would this public published URL feature work with yours as well?

ealmonte32, 2019-02-21:
@hycsa and @outtasync 

Can you log into the Pi via ssh? using maybe VNC viewer so that you can use your keyboard?

If you guys give me the exact URL for these websites I can test on my end to see if i figure out a workaround since like vpetersson said - it is a human interface device limitation when plugged into Pi.
Workarounds I can think of include passing login info via URL, or some script that leaves the login info saved in the uzbl session data store like autofill + autosubmit when reaches specific URL..

hycsa, 2019-02-25:
@ealmonte32 its a sharepoint for a school. The thing we want to do is publish the news articles thats on sharepoint, to do this you have to validate yourself. I could give you the link but I cant publish the school name.. : https://schoolname.sharepoint.com/sites/intranet. When I put this link in Screenly it sends me to the login page which I cannot type to validate myself.

ealmonte32, 2019-02-25:
@hycsa - Sorry I dont have sharepoint in order for me to find a workaround, and trying to script something for uzbl to parse your login info may not work.. but I read that you are allowed to have shareable links that dont require sign-in, thus you would sign in from your computer, publish the news articles you want to show on screenly, then copy that URL to the Assets page instead of the page you are saying it asks for sign-in..

Reference:
`
https://sharepoint.uservoice.com/forums/329214-sites-and-collaboration/suggestions/13541985-external-folder-sharing-with-without-sign-in#{toggle_previous_statuses}
`

hycsa, 2019-02-27:
@ealmonte32 Your method with the VNC viewer, if I use the VNC viewer does this also make it possible to use it for multiple screens?

ealmonte32, 2019-02-27:
Yes, I just tested it.

On the Screenly Pi, do this:

Install VNC server:
`sudo apt-get install realvnc-vnc-server`

Enabling VNC Server from the command line:
`sudo raspi-config`

Now, enable VNC Server by doing the following:
Navigate to Interfacing Options.
Scroll down and select VNC > Yes.

Let me know if that works for you.

vpetersson, 2019-02-27:
Make sure to not forget to set a password for VNC.

ealmonte32, 2019-02-27:
> Make sure to not forget to set a password for VNC.

:)

Yeah, type:
`vncpasswd`

ealmonte32, 2019-03-04:
@hycsa - if this solves your problem can we close this issue? thx.

Brian78912, 2019-03-10:
Hi trying to accomplish the same, with no luck. 
Have tested vnc suggestion but it do not take any keyboard input, in the login form.

[image]

ealmonte32, 2019-03-12:
> Hi trying to accomplish the same, with no luck.
> Have tested vnc suggestion but it do not take any keyboard input, in the login form.
> 
> I have also testet the 'Experimental' version, no keyboard when connected to USB. VNC seems not to work.

Hmm.. I don't want to say impossible because I know some other service can disable HID input, but that's a weird one.
Can you describe a bit what you did?
Ran the commands I posted above to install vnc server? and connect but can't get mouse to work?
Did you follow instructions from here:
`https://www.raspberrypi.org/documentation/remote-access/vnc/`
Are you running the production(standard) version or development or experimental?

vpetersson, 2019-03-12:
@ealmonte32 note that in the experimental branch, i don't think VNC will work, as it's running in framebuffer. It should however work in the master/produciton branch.

For the experimental branch, there is actually a built-in VNC functionality built into QT that we should be able to expose.

ealmonte32, 2019-03-13:
@vpetersson 
What if he uses http://manpages.ubuntu.com/manpages/bionic/man1/directvnc.1.html

hycsa, 2019-03-13:
Hello, sorry for the late response I was on a holiday and couldnt reach the office where the Pi is at. I've followed your steps for the VNC Viewer and like @Brian78912 said, no success. Are there any other options available to do? My mouse still works but I just cant type which is the problem. If I manage to log in with my credentials my project would be literally done. So close yet so far :(.

ealmonte32, 2019-03-13:
> Hello, sorry for the late response I was on a holiday and couldnt reach the office where the Pi is at. I've followed your steps for the VNC Viewer and like @Brian78912 said, no success. Are there [...]

@hycsa 
Welcome back.
Did you read my comment about directvnc ?

Brian78912, 2019-03-14:
Thanks for helping :)

@ealmonte32  standard and experimental version testet. 
I testet experimental with directvnc, no luck.  (http://manpages.ubuntu.com/manpages/bionic/man1/directvnc.1.html)

Gives this output i console, 

[image]

Regarding https://www.raspberrypi.org/documentation/remote-access/vnc/ - yes i did. 

@vpetersson  sound good, i can test it but don't know how.

ealmonte32, 2019-03-15:
@Brian78912 

I just got some extra Pi for testing so I will be able to do a more exact test that resembles what you are doing.

Can you please confirm:
1. Raspberry model and Debian/Raspbian version:
```
cat /sys/firmware/devicetree/base/model
cat /etc/debian_version
```
2. Screenly version/branch you want me to test on:
`Production/Development/Experimental`

3. VNC Viewer you are using on your computer that wants to remote control the Pi:
`VNC Viewer from RealVNC, or ?`

Brian78912, 2019-03-17:
Cool :)

1: Raspberry Pi 3 Model B Rev 1.2. Debian version: 9.8
2: Production is fine
3: REal VNC Viewer

ealmonte32, 2019-03-20:
> @ealmonte32 Cool :)
> 
> 1: Raspberry Pi 3 Model B Rev 1.2. Debian version: 9.8
> 2: Production is fine
> 3: REal VNC Viewer

Thanks for info, was testing some today.
Directvnc did give me keyboard mapping error as well, very old bug and did not want to waste time trying to apply the "supposed" fix for it.
Not many framebuffer supporting VNC packages out there, but I think I can do it with x11vnc.
Will test tomorrow and let you know.
`x11vnc - VNC server to allow remote access to an existing X session`
This is why it should work with that because as far as I know screenly-viewer is already using X..

_Ref: https://wiki.debian.org/VNCviewer_

ealmonte32, 2019-03-25:
An update:

@Brian78912 

So without changing anything, I was not able to type on the page that screenly shows as the asset, but since mouse works, I was messing around and right-clicked on the page and then selected Inspect Element, just to test different things, this inspect element mode as we know from all other browsers (in firefox is called developer options) shows the backend of the page loaded, and it was actually allowing me to type, so I know that uzbl has insert mode and command mode, but I am not that familiar with overwriting the current asset being displayed and somehow changing the display from command mode to insert mode (pressing ctrl+i did not work)..

I tried different methods with different vnc servers that support X term, and I ran into this XKEYBOARD extension requirement, but then I asked myself if that was indeed required for the keyboard to function, why was I able to use the keyboard without any extension or different vnc package on the uzbl loaded page when on "inspect element" area..

I have to put this on the back-burner for now since there are other things in need of troubleshooting,
I believe rusko is working on some easier form of keyboard input for loaded assets, so unless someone else has found a way, sorry I dont have better news yet.

Brian78912, 2019-03-25:
@ealmonte32 well thanks for trying :)

ealmonte32, 2019-03-27:
@Brian78912 @hycsa @outtasync 

Guys, I got the keyboard input to work by loading the default uzbl config into the config-screenly, but you need to do some manual work to only set the keyboard stuff and remove everything else..
I would suggest you edit the config file in your regular desktop environment, use something like Filezilla to connect to the Screenly Pi and navigate to that location and either direct edit or copy file over to your system, edit the file, then copy it back.

```
sudo cat ./.config/uzbl/config >> ./.config/uzbl/config-screenly
```
then go into the config-screenly file, find and make changes, like the first change you need to do is to
```
set show_status = 0
```
 by searching for the that line which appears twice since the config-screenly default already has that line..
also, delete the bottom part where it makes the homepage uri the uzbl one, so on the last part of the file find the "set home page" part and delete that.. example of what to delete:
```
# Set the "home" page.
set uri = uzbl.org/doesitwork/@COMMIT
```

also, i dont have too much time right now to play around with all config stuff, but there are tons of commands that do not need to be copied over but since i am not going to go through each one, just making these few changes and restarting the screenly service will let you type into the page..

the default home will also need to be deleted because then local assets wont load..

i will attach my config-screenly file here so you guys can use as starting point..

**Uploaded assets such as pictures will need one full cycle of loading and you will probably see a black screen first, but when the assets cycle/loop back again they should display, this is probably because of cache and change of directory**

**this is very temporary solution because this default uzbl configuration file might break some things..**

Edited config-screenly attached as .txt for upload purposes, just remove the extension and leave extensionless when copying into your /uzbl/ directory...
[config-screenly.txt](https://github.com/Screenly/screenly-ose/files/3011727/config-screenly.txt)

ealmonte32, 2019-03-29:
@hycsa 
Since this is working forme, I would want to close the issue, so give my config-screenly file a try, I've been using it without problems all this time, i am able to interact with loaded website page just fine.. any issues you encounter after using my config-screenly file just post back at the forums instead of here since it wont be an issue but more of a side effect that can be resolved by removing certain lines from uzbl config file.

As I said, this should take care of it:
```
curl https://raw.githubusercontent.com/ealmonte32/screenly-ose/master/templates/config-screenly > ./.config/uzbl/config-screenly
```
then `sudo systemctl restart screenly-viewer.service`

closed 2019-06-11 by stale[bot]