Screenly/Anthias #1031: URL Assets restarts the browser

denfau, 2019-02-11:
**Overview of the Issue**
Hi, I have a few Web assets that are not loading properly.
One is a link to a specific Instagram Post, the other link is to a Widget page that should display a class calendar.
The links work without issues in Firefox, Edge, IE.

**Reproduction Steps**

* Press Add Asset
* Enter the Asset URL "https://www.gorendezvous.com/BookingWidget/?companyId=108450&classesOnly=1"
* Display shows a page showing "loading/connecting"
* Display becomes blank and reverts to startup splash screen
* I can turn that specific asset off, and the viewer will restart to show other assets (JPG files)

### Environment

* Raspberry Pi Hardware Version: Raspberry Pi 3 B
* Screenly OSE Version: Experimental Branch

Thank you.

qbicdesign, 2019-02-21:
I can confirm this same problem with image_2018-11-23-Screenly-OSE-lite

Screenly's own widget URLs load fine. Most other URLs cause the browser to crash.

I have to uninstall and use a previous version as this is unuseable.

vpetersson, 2019-02-21:
This is very strange, because we haven't really changed anything related to this in Screenly OSE (other than in the experimental branch). Can you try to run an upgrade to the latest developer release? That should also upgrade the system packages.

qbicdesign, 2019-02-21:
How would I do that exactly? can't find instructions. If you mean just run the upgrade script, $ bash <(curl -sL https://www.screenly.io/install-ose.sh)   then yes I can do that. Anything more you'll need to give me pointers.

vpetersson, 2019-02-21:
You're spot on. That's it.

qbicdesign, 2019-02-21:
So  - a little conflict in terminology.
The upgrade script invites me to use the experimental branch, with new browser and Docker.
is developer realse and experimental branch one and the same thing? If not what should I answer at the various prompts I'm gonna get?

qbicdesign, 2019-02-21:
I seem to be stuck on TASK [system : Perform system upgrade]
Screenly itself is still working and I can still access web console etc. Just the script seems to be doing nothing. Its been like 20 mins since it last moved. Is that normal?

qbicdesign, 2019-02-21:
go figure - as soon as I posted it woke  up :)

vpetersson, 2019-02-21:
@qbicdesign that's a heavy operation. Depending on your interenet connection, SD card and Pi, 20 minutes+ isn't unexpected.

qbicdesign, 2019-02-21:
So... 

After upgrading to dev branch, instead of crashing it seems the Screenly just skips assets that it (for some reason) thinks are not available.

The clock widget loads, as does one of our internal URLs

The weather widget (which is customised for our locale) won't load:
Feb 21 16:15:24 raspberrypi python[478]: Asset Ilm at https://weather.srly.io?lat=59.42816382263877&lng=24.533778739874272&lang=et&24h=1&wind_speed=1 is not available, skipping.

And my Google slides presntation is also skipped:
Feb 21 16:21:00 raspberrypi python[478]: Asset pohiesitlus at https://docs.google.com/presentation/d/e/[xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx]/pub?start=true&loop=true&delayms=10000&slide=id.g47be5881ff_1_8 is not available, skipping.

qbicdesign, 2019-02-21:
I should add that I'm currently running on Raspberry Pi 2 model B+

denfau, 2019-02-21:
An update to this issue.
When I put only two assets which are URL, the application gives errors, related to utf-8.
The page has accented characters (french).

The first URL is loaded to set the Widget language to french:
https://www.gorendezvous.com/fr/homepage/108450/

The second URL is the call to the Widget:
http://www.gorendezvous.com/BookingWidget/?companyId=108450&classesOnly=1

I just updated the RPi to the latest development branch.

See attached log file.

[Logs1.txt](https://github.com/Screenly/screenly-ose/files/2889858/Logs1.txt)

denfau, 2019-02-21:
About the other issue with an Instagram URL Asset, the Instagram post page is shown correctly, then the next Asset, a JPEG image, is shown, but the browser freezes at that point and the last image stays on.

The logs indicate:
> current url is file:///tmp/screenly_html/black_page.html

The page is not black, the last image is always on, and the viewer never restarts.

See logs.
[logs2.txt](https://github.com/Screenly/screenly-ose/files/2889945/logs2.txt)

denfau, 2019-02-21:
I closed this by mistake... re-opening...

ealmonte32, 2019-02-21:
Hmmm.. I was also getting the black page, but I removed all assets, then added the following assets:

https://www.gorendezvous.com
then
https://www.gorendezvous.com/BookingWidget/
then
https://www.gorendezvous.com/BookingWidget/?companyId=108450
then
https://www.gorendezvous.com/BookingWidget/?companyId=108450&classesOnly=1

Just for testing purposes to see at what point it stopped loading, and I can't seem to find the reason as to why now it started working..

@denfau - are you still experiencing this issue?

can you tell me what version of uzbl you get when you do the following:
`sudo apt-get install uzbl`

example:
`uzbl is already the newest version (0.0.0~git.20120514-1.2+b2).`

Do you happen to know what version of libwebkitgtk and libjavascriptcoregtk you have? just wondering.

[image]

denfau, 2019-02-22:
@ealmonte32 
This is what I get for uzbl:
uzbl is already the newest version (0.0.0~git.20120514-1.2+b2).

As for libwebkitgtk and libjavascriptcoregtk, I do not know what versions...

I will be able to provide additional info only in a few days

ealmonte32, 2019-02-22:
> @ealmonte32
> This is what I get for uzbl:
> uzbl is already the newest version (0.0.0~git.20120514-1.2+b2).
> 
> As for libwebkitgtk and libjavascriptcoregtk, I do not know what versions...
> 
> I will be able to provide additional info only in a few days

ok no problem, i just want to always check first to see if something that works for me and not for someone else has to do with some version of a library or something else..

here are the commands sorry for not posting it before:
`dpkg -l | grep libweb`
`dpkg -l | grep libjava`

ealmonte32, 2019-02-22:
@denfau - also when you can, update javascript and reinstall uzbl web browser, takes few seconds.

commands:

`sudo apt-get install javascript-common`
`sudo apt-get install --reinstall uzbl`

ealmonte32, 2019-02-23:
@denfau @qbicdesign - this is probably a guaranteed fix for these websites..

This is a sort of not very well liked setting out there in the python world because it sets a global default that may not work with some websites, but to fix this edit the following file:
`sudo nano ./screenly/viewer.py`

then somewhere on the top add the following:

>#encode all utf-8
import sys
reload(sys)
sys.setdefaultencoding('utf8')

ctrl+x to save, press y, then enter..

then you can just reload screenly viewer like this:
`sudo systemctl restart screenly-viewer.service`

qbicdesign, 2019-02-26:
Seems to have fixed it for me.
Restarting the viewer just made Screenly get stuck on its current asset.
After a reboot all the assets including URLs are showing correctly.
Lets see if it sticks.

ealmonte32, 2019-02-26:
> Seems to have fixed it for me.
> Restarting the viewer just made Screenly get stuck on its current asset.
> After a reboot all the assets including URLs are showing correctly.
> Lets see if it sticks.

We're working on a better fix to deal with URLs containing unicode and non-ascii characters through encoding.

rusko124, 2019-03-01:
@denfau Hi!
The problem you are talking about rather belongs to uzbl(Screenly-OSE based on it at the moment).
`/usr/bin/uzbl-browser --config=- --uri='https://www.gorendezvous.com/BookingWidget/?companyId=108450&classesOnly=1' --print-events
`
Uzbl crashed in my case.
I can advise you to try to switch the Screenly to experimental branch as it is based on QtWebKit.

qbicdesign, 2019-03-04:
@ealmonte32 After applying your fix, and now after a reboot, no URL assets at all are showing, it just skips to show only video assets

qbicdesign, 2019-03-04:
I just don't get it tho, cos this has worked flawlessly in previous versions. 
My other TVs are running 2018-05-04-Screenly_OSE_4GB.img and no problems at all...

ealmonte32, 2019-03-04:
The setting of system default to utf8 made it not show any assets at all?
Ugh..
Can we see what the error is now?
Running that journalctl command again should tell us something..

And about working flawlessly before , you know lots of changes have happened over time, and raspbian and the uzbl browser itself have changed too, but yes it's something between Screenly and the browser uzbl most likely.. it's hard to isolate it without debugging it.

qbicdesign, 2019-03-04:
Weird that the log is showing me that the URL isn't skipped on all, but as far as I have seen so far, they don't show at all.
The only URL noted as skipped is weather
Mar 04 13:57:59 raspberrypi python[514]: Asset Ilm at https://weather.srly.io?lat=59.42816382263877&lng=24.533778739874272&lang=et&24h=1&wind_speed=1 is not available, skipping.

qbicdesign, 2019-03-04:
@ealmonte32 what journalctl command are you talking about?

ealmonte32, 2019-03-04:
@qbicdesign 

So I left two screenly Pi's over the weekend running the same assets/URLs, so far one of them continued and did not stop at all, but I removed the weather.srly.io asset on both, and I also removed the boardclock one, both gave the same Unable to load error with those on. I have google slides running on both and that never gives the Unable to load error.

But, now I got an "Unable to load: Message corrupt" on one of the Pi's, but this url that gave the corrupt message is also on the other Pi as well and that one did not give such error, so it is very hard to isolate the issue. As you know when figuring something out we need to isolate piece by piece and see what works and what doesn't, but since this is a random works-now-doesn't-later type issue, it must be with uzbl browser.. the screenly viewer has not changed so it must be the process that is retrieving the URLs, aka uzbl browser.. 

The command to see the logs of what is going on in the background processes is called journalctl, if you type that in the console of the Pi it should give you lots of messages, but if you want to post here, you can either echo all to a text file and upload or paste the filtered log here by typing:
`journalctl | grep "Mar 04"`
That for example will display only logs from march 4th..
Also, when you post lots of logs here you should use the markdown ``` before the pasting of output and after it so that it makes it easier to read.

ealmonte32, 2019-03-13:
@denfau @qbicdesign 

Just to let you guys know that I am able to load the page now every time without a problem in terms of browser restarting, (maybe rusko fixed the unicode in uri titles for uzbl stuff i havent checked the changes in the viewer code), but read my comment: https://github.com/Screenly/screenly-ose/issues/806#issuecomment-472508436

nwdigital, 2019-03-14:
> @qbicdesign
>
> So I left two screenly Pi's over the weekend running the same assets/URLs, so far one of them continued and did not stop at all, but I removed the weather.srly.io asset on both, and I [...]

Hi, I'm the owner/create of boardclock.com. I was doing some searching on the web and saw your thread describing the issue with the TLS problem causing screenly to freeze. I just want to let you know, I and another friend of mine both had the same issue with boardclock, not sure what the issue is as the SSL certs should be fine. But, I can tell you this, I reinstalled screenly and chose the dev version during install and I have not had an issue since.

If there is anything I can help with please let me know.

ealmonte32, 2019-03-14:
@nwdigital 

Correct, it is not an issue with boardclock.com , it just happened mostly on boardclock and screenly URL because during testing phase those were the most commonly used URLs added to the assets, thus it just happened to be that it got mentioned the most because when assets were on google.com domain it would not give TLS handshake error.
But I have been getting error on all other sites as well with production version.

This TLS issue actually belongs on another post, it sort of got intertwined by mistake into this because some users were getting both, TLS error and browser crashing because of uzbl uri unicode errors.

_P.S. boardclock is still displaying daylight savings time =]_

[image]

nwdigital, 2019-03-14:
> @nwdigital
>
> Correct, it is not an issue with boardclock.com , it just happened mostly on boardclock and screenly URL because during testing phase those were the most commonly used URLs added to the [...]

Thanks for following up with me. As for daylight savings time, the old way I was getting time was from the device it was being displayed on. I have since added a new feature to get UTC time and convert it to whatever timezone you choose, feel free to login to https://boardclock.com and adjust the setting according in the Timezone Selection section.

[image]

qbicdesign, 2019-03-21:
A little update from me.

I made a new install of Screenly for a new TV in our school. During initial install I had the Raspberry on same VLAN as all our desktop PCs, I used OSE version 2018-05-04-Screenly_OSE_4GB.img
After I moved the Raspberry onto another VLAN (where all my other Screenly raspberries are) the external URL assets were again being skipped.  After a little investigation we discovered that the second VLAN was using our ISPs DNS servers. 
I added Google as secondary DNS 8.8.8.8, and hey presto, the external URLs started working again after a reboot.

So this does actually relate to my earlier suggestion :
> There is no reason I can see that these URLs should be unavailable.
> Can it maybe some kinda latency issue? That the browser quits looking if page load latency is beyond a certain number of milliseconds?

vpetersson, 2019-03-21:
@qbicdesign 

> Is there anything that can be done programatically to make Screenly's DNS queries a little more patient?

Interesting find. We don't really do anything fancy with networking tho. Just using the built-in functionalities really.

qbicdesign, 2019-03-21:
Having said that, I'm now finding that the Screenly weather and clock widget pages don't always load fully. 
Sometimes I get black screen
Sometimes I get just the background
Sometimes I get the fully loaded widget

I may try turning of the ISP DNS completely and see if that makes a difference.

I'm seeing these errors in the logs
- python[614]: InsecureRequestWarning)
- python[614]: /usr/local/lib/python2.7/dist-packages/requests/packages/urllib3/connectionpool.py:821: InsecureRequestWarning: Unverified HTTPS request is being made. Adding certificate verification is strongly advised. See: https://urllib3.readthedocs.org/en/latest/security.html
Not sure if this is related at all

ealmonte32, 2019-03-25:
> Having said that, I'm now finding that the Screenly weather and clock widget pages don't always load fully. Different results on different TVs yet all running identical versions.
> Sometimes I get black screen
> Sometimes I get just the background
> Sometimes I get the fully loaded widget

This is not python/screenly related, but mostly uzbl and libwebkit because it is what handles the content that loads on the screen.
___

> I may try turning of the ISP DNS completely and see if that makes a difference.
>
> I'm seeing these errors in the logs
>
> * python[614]: InsecureRequestWarning)
> * python[614]: [...]

That warning is related to the urllib package as you can read from the URL it gives you, it doesn't prevent the page from loading the contents. Also, that is another topic/issue, you can search the forums and Issues for the keyword "tls error" and you can read all things we tried to do to isolate and pinpoint the problem, and only recommendation I have as the best workaround is to create a Google Sites page and literally just use the insert URL feature and put the website you are trying to reach into it, you will be connecting to sites.google.com instead of the other pages, which has not given me the TLS error.
If you want to read about this workaround: https://github.com/Screenly/screenly-ose/issues/974#issuecomment-471852566
___
Ok, last but not least:
Remember that when you flash the SD card using the screenly image, it was built almost a year ago as you can see from the build date. Lots of changes to shared libraries and packages have happened since then, thus I recommend that when you want to set up screenly, even though it takes more time because it is not preinstalled, I would suggest you take a raspbian lite image (https://www.raspberrypi.org/downloads/raspbian/) 
and properly set up the localisation options (utf-8 should be default), hostname, network, etc, and then use the curl method to install screenly
`bash <(curl -sL https://www.screenly.io/install-ose.sh)`

If I were you I would add the NTP pool address of your corresponding timezone to the following file:
`sudo nano /etc/systemd/timesyncd.conf`
where it says [Time], for example I use `NTP=0.north-america.pool.ntp.org`
just to keep the Pi time synced properly.

After all has been installed and setup, etc, I highly recommend updating all libraries/packages:
`sudo apt update && sudo apt full-upgrade -y`
then the optional
`sudo rpi-update`
If you need to read any firmware related changes: https://github.com/raspberrypi/firmware

denfau, 2019-03-26:
@ealmonte32 

I did a fresh install following your suggestions.

I have setup only two URL assets for now, Screenly Weather and Screenly Clock.
I have chosen the Development branch when I installed Screenly.

I will monitor over the next 24h, and let you know how things go.

Thanks to all for helping.

ealmonte32, 2019-03-26:
> @ealmonte32
>
> I did a fresh install following your suggestions.
>
> I have setup only two URL assets for now, Screenly Weather and Screenly Clock.
> I have chosen the Development branch when I installed [...]

No prob.
Looking forward to hearing how it goes.
When you added the two URLs, you did direct URL or embedded into a google sites URL?
I just want to make sure everything is noted down exactly as you are doing.

denfau, 2019-03-29:
The two Screenly web assets are still running without any issues.
They are direct URL, not embedded.

I will try a few Instagram URL assets for a few days, which I had trouble with...
Such as https://www.instagram.com/p/BvZZZz5BV8j/

ealmonte32, 2019-04-01:
@denfau 
Any updates regarding the assets and browser issues?
I have narrowed down that depending which URL you add as asset, that is what causes TLS error..
I am still doing research on certificates on the Pi, certificate differences on the sites which cause the TLS error and the ones that dont by checking certificate info using site such as https://ssltools.digicert.com
 
To stop getting InsecureRequestWarning, simply do this:
[image]

_(all other suggestions to suppress this warning should be ignored)_

denfau, 2019-04-02:
@ealmonte32 
The two Instagram assets are still showing without any issues.

I still see this warning in the logs, but it does not stop the viewer.

> /usr/local/lib/python2.7/dist-packages /urllib3/connectionpool.py:847: InsecureRequestWarning: Unverified HTTPS request is being made. Adding certificate verification is strongly advised. See: https:                                                                                //urllib3.readthedocs.io/en/latest/advanced-usage.html#ssl-warnings

Thank you for the research and help.

ealmonte32, 2019-04-02:
> The two Instagram assets are still showing without any issues.
>
> I still see this warning in the logs, but it does not stop the viewer.
>
>
> Thank you for the research and help.

@denfau 
I forgot to tell you, the verify = True is done in the file located in
`sudo nano /home/pi/screenly/lib/utils.py`
As far as I know, this will stop that unverified request warning.

rusko124, 2019-04-19:
@vpetersson 
So https://www.gorendezvous.com/BookingWidget/?companyId=108450&classesOnly=1 url equally does not work as on 2018-05-04-Screenly_OSE_4GB and image_2018-11-23-Screenly-OSE-lite.

This problem concerns UZBL browser and not Screenly. 

`/usr/bin/uzbl-browser --config=- --uri=https://www.gorendezvous.com/BookingWidget/?companyId=108450&classesOnly=1 --print-events`
Command doesn't work, browser crash.

But we also have some issues related to a similar error. These were solved here #1039

Thus I suggest move this issue here https://github.com/uzbl/uzbl/issues and close it.

vpetersson, 2019-04-20:
Ok let's move away from this as soon as possible then.

closed 2019-04-26 by vpetersson