Screenly/Anthias #1032: Balena PI Zero having issues with cec

seesoe, 2019-02-19:
#### Overview of the Issue
Trying to add some cron jobs to control my tv's on and off with CEC. I found the idea from here
https://www.screenly.io/blog/2017/07/02/how-to-automatically-turn-off-and-on-your-monitor-from-your-raspberry-pi/

However after installing (and updating) I get the following error when trying to execute a command
```
libvcos.so: cannot open shared object file: No such file or directory
Cannot load libcec.so
```

#### Reproduction Steps
```
echo 'on 0' | cec-client -s -d 1
```

### Environment

* Raspberry Pi Hardware Version: Zero
* Screenly OSE Version: latest

vpetersson, 2019-02-19:
@seesoe Looks like libcec is broken. You might need to compile it as part of the docker build. Take a look at [these build instructions](https://github.com/Pulse-Eight/libcec/blob/master/docs/README.linux.md).

seesoe, 2019-02-20:
@vpetersson thanks for the response. I started trying to do so in the docker container but a believe it might be wrong to do it there and it needs to be on the host os? I'm having trouble finding many of these paths. 
`cmake -DRPI_INCLUDE_DIR=/opt/vc/include -DRPI_LIB_DIR=/opt/vc/lib ..`

`/opt/` dir is empty.

```
cmake -DCMAKE_TOOLCHAIN_FILE=../cmake/CrossCompile.cmake \
      -DXCOMPILE_BASE_PATH=/path/to/tools/arm-bcm2708/arm-bcm2708hardfp-linux-gnueabi \
      -DXCOMPILE_LIB_PATH=/path/to/firmware/hardfp/opt/vc/lib \
      -DRPI_INCLUDE_DIR=/path/to/firmware/hardfp/opt/vc/include \
      -DRPI_LIB_DIR=/path/to/firmware/hardfp/opt/vc/lib \
      ..
```
from https://github.com/Pulse-Eight/libcec/blob/master/docs/README.raspberrypi.md

vpetersson, 2019-02-20:
> I started trying to do so in the docker container but a believe it might be wrong to do it there and it needs to be on the host os?

No, you should e able to do it within the container on Balena since they run in privileged mode.

I honestly haven't tried this and unfortunately can't look at it now. I would however imagine that you might need to include some other library if that's the error you're getting.

Also, I'd recommend that you modify the `Dockerfile` directly rather than doing it inside the container. You might also find some help related to this on the Balena forum, as they tend to be pretty active.

closed 2020-01-17 by stale[bot]