Screenly/Anthias #1033: Database Error

l888, 2019-02-20:
Hi

I'm new to your great software.

I have the latest version (Screenly OSE) how do i transfer 40+ JPG pictures all at once without getting a Database Error when i drag and drop the JPG's into the upload asset folder ?
also i need them all to be set to Play forever can this be done to them all in one go? 
Thanks

Lee

vpetersson, 2019-02-21:
@l888 My best guess is that you have a broken SD card. Try replacing it and start over.

rusko124, 2019-03-01:
Hi @l888 !
Can you try update to the development version? It contains some edits to work with large amounts of assets in the database.

ealmonte32, 2019-03-25:
@l888 

I uploaded 40 images in experimental version just fine:

[image]

To make the asset play "Forever" as default, you need to edit the server.py file:
`sudo nano /home/pi/screenly/server.py`

and find the line that mentions the end date and change the last bit like I did adding the
", year=+9999" part:
```
    # parse date via python-dateutil and remove timezone info
    asset['start_date'] = date_parser.parse(get('start_date')).replace(tzinfo=None)
    asset['end_date'] = date_parser.parse(get('end_date')).replace(tzinfo=None, year=+9999)

    return asset

# api view decorator. handles errors
def api_response(view):
    @wraps(view)
    def api_view(*args, **kwargs):
        try:
            return view(*args, **kwargs)
        except Exception as e:
            traceback.print_exc()
            return api_error(unicode(e))
    return api_view
```

closed 2019-05-31 by stale[bot]