Screenly/Anthias #1034: screenly-ose version for one display only

Jom-o, 2019-02-21:
Hi guys

wanna take the screenly ose version for one display only.

BUT: The pi gives me this adress: http://2003:de:b72e:9d93:7006:781b:e1d2:fd02

There can’t change the pi-content.

Thanks for your help

* Raspberry Pi Hardware Version: 3
* Screenly OSE Version: 2018-01-19-Screenly_OSE_4GB

vpetersson, 2019-02-21:
Looks like the device getting an IPv6 IP address. I don't think we have tested it very closely for such environment. Is your network IPv6 only?

Jom-o, 2019-02-21:
no - pi becomes an IPv4 - thanks

vpetersson, 2019-02-21:
Try logging into the device. Something is likely wrong with the network. Once you're logged in, please provide the output of `ifconfig` or `ip a show`.

Jom-o, 2019-02-21:
the problem: pi run from the SD-Card - is loading - after this -now the screen shows maybe 30 seconds this web site for the browser. After this, there are now feed picture and weather from screenly. that's all - please tell me, where i can show you the ifconfig ... thx

vpetersson, 2019-02-22:
@Jom-o Sounds like you need some assistant getting up and running with the basic Linux stuff. I suggest posting on the [forum](https://forums.screenly.io/c/screenly-ose) for that part.

Jom-o, 2019-02-22:
OK - thank you. Yesterday, i load the 'First, flash the SD card and install Raspbian Lite. Instructions are available here' - the command $ bash did my pi not find. I will post my next question to this new forum.

Jom-o, 2019-02-23:
Thanks - i have found the problem. it was me. - thank for your support. br j

Hypnoblimp, 2019-03-14:
I'd be grateful for some pointers having encountered an issue similar to that raised by Jom-o above. I've used earlier versions of Screenly OSE without any problem. Downloaded Screenly 2018-11-23 and used Balena Etcher to burn the image to an 8GB SD card. Used a Pi 3 Model B V1.2 and Screenly booted in the usual way, except that when reporting the Manager IP address it gave this as:
    http://2a00:23c4:, etc
i.e., a series of 8 4-digit hex codes. 
I've tried burning using the extracted img and zip files with 3 different SD cards, all with the same result. Burning an earlier version of Screenly works fine and reports the IP address in the usual format

2018-11-23 works well in every other respect: both my router and ifconfig (and `ip a show`) all show the same IP address in the expected range which presents the usual management screen. I can upload and address assets in the usual way.

I'd be grateful for any suggestions as I'm installing signage for a charity

Jom-o, 2019-03-14:
Hi Hypnoblimp

please check all the things you do.

first - take image with balena etcher to your sd-card
second - take time, to install the pi - you must fist really clear declare 
your password and the shell activation 

You'll find all the parameters under forum.

Otherwise - if i can help - let me know.

You can call me by Skype or otherwise

br jo

vpetersson, 2019-03-14:
@Hypnoblimp - that sounds like an IPv6 address. Chances are the device fails go get an IPv4 address.

Hypnoblimp, 2019-03-14:
Ok - thanks both. I'll check these out

Hypnoblimp

Hypnoblimp, 2019-03-15:
Still not making much progress with this issue. I've confirmed with `ifconfig` that IPv6 is operative for both eth0 and wlan0. I edited    /etc/sysctl.conf
adding the line: `net.ipv6.conf.all.disable_ipv6 = 1`. `ifconfig` now reports:

        inet 192.168.1.162  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 2a00:23c4:398c:be00:e26c:42b6:326:5cee  prefixlen 64  scopeid 0x0<global>
        inet6 fdaa:bbcc:ddee:0:12c0:7b08:23ee:ab27  prefixlen 64  scopeid 0x0<global>
        inet6 fe80::2026:770b:9320:a197  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:21:36:4a  txqueuelen 1000  (Ethernet)
        RX packets 958  bytes 836949 (817.3 KiB)
        RX errors 0  dropped 1  overruns 0  frame 0
        TX packets 665  bytes 76261 (74.4 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

Any further ideas much appreciated. The first IPv6 address is the same as that on the splash screen

Many thanks

Hypnoblimp

vpetersson, 2019-03-15:
@Hypnoblimp looks like you have a valid IPv4 indeed. @rusko124 any ideas?

rusko124, 2019-03-19:
Hi @Hypnoblimp!
Could you provide some outputs, please?
- `ifconfig` without any edits in device system
- `python -c "import netifaces;print(netifaces.gateways())"`

Hypnoblimp, 2019-03-19:
Hi @rusko124 

Thank you for getting back to me. I've commented out the change I made to the `/etc/sysctl.conf` file and rebooted. `ifconfig` brings up:

```
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.162  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 2a00:23c4:398c:be00:e26c:42b6:326:5cee  prefixlen 64  scopeid 0x0<global>
        inet6 fdaa:bbcc:ddee:0:12c0:7b08:23ee:ab27  prefixlen 64  scopeid 0x0<global>
        inet6 fe80::2026:770b:9320:a197  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:21:36:4a  txqueuelen 1000  (Ethernet)
        RX packets 12982  bytes 18609909 (17.7 MiB)
        RX errors 0  dropped 2  overruns 0  frame 0
        TX packets 7431  bytes 635691 (620.7 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 2583  bytes 308306 (301.0 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 2583  bytes 308306 (301.0 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether ba:bf:ee:b7:7f:35  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

```
and the short python script brings up:

```
{'default': {2: ('192.168.1.254', 'eth0'), 10: ('fe80::42c7:29ff:fe0e:b3b5', 'eth0')}, 
2: [('192.168.1.254', 'eth0', True)], 
10: [('fe80::42c7:29ff:fe0e:b3b5', 'eth0', True)]}

```
Many thanks

Hypnoblimp

rusko124, 2019-03-22:
@Hypnoblimp @Jom-o 
If you still have a problem, could you upgrade to the development version and tell if this helped you?

Jom-o, 2019-03-22:
@rusko124 thanks - there are no problems in my project at the moment  - br j

vpetersson, 2019-03-23:
Great! Closing out issue.

closed 2019-03-23 by vpetersson

Hypnoblimp, 2019-03-25:
Thank you

I'm unsure where to find the development version as I'm new to GitHub. As posted previously, Screenly OSE version 2018-05-04 works fine and displays an IPv4 address on the splash screen.

Should I just revert to using the older version? 

Thanks

Hypnoblimp

Hypnoblimp, 2019-03-25:
@rusko124 

Further to my comment above, I've upgraded the system using 
` $ bash <(curl -sL https://www.screenly.io/install-ose.sh)` and on reboot the IP address appeared in IPv4 format
So, thank you for that advice

Hyonoblimp