Screenly/Anthias #1036: Screenly OSE freezing up.

trustees629, 2019-02-23:
Ok, I'm having a problem with my Screenly OSE freezing after updating. I had mentioned it in issue #1012 but couldn't duplicate. Today, I just updated to the latest release 555053e and it's frozen again. I am attaching a journalctl | grep "Feb 23" with the hope that I can help get this narrowed down. 
[ose-frozen.txt](https://github.com/Screenly/screenly-ose/files/2897112/ose-frozen.txt)

vpetersson, 2019-02-24:
@trustees629 - Could you try swapping out the SD card? I see a timout in the logs, and that can often happen if the SD card is broken or on the verge of breaking.

trustees629, 2019-02-24:
@vpetersson - Sure, I can do that. But this has been running non stop for the past 3 weeks with 57dbde3 and didn't start doing it again until I updated to the latest release.

rusko124, 2019-03-01:
Hi @trustees629 !
Some edits have just been made(development version). Could you please update and let us know if it helped you?

trustees629, 2019-03-01:
Hi @rusko124 , 

I updated to the latest master release, f16d953, and I'm still having the problems with it freezing. I have two systems running this and it's happening for both. I'm wondering if there's something off with my Balena application. I'm going to take one of them down, create a new application and go from there.

ealmonte32, 2019-03-01:
@trustees629 - when was the last time you did a full system upgrade of the Pi and all the packages? your log shows X.org version from 2016, my screenly one shows ver 2017-03-02, I am not saying this is at all the issue, but makes me wonder about the packages you have and their out of date versions..

Can you please run:
`sudo apt-get update && sudo apt-get upgrade`

Also, if possible to know, is the district2.png the same file that it always freezes on?

trustees629, 2019-03-01:
@ealmonte32  - This is running through the Balena cloud. However, I did do an update yesterday to the latest Balena release. I don't appear to be able to run the command you suggested in the terminal window through Balena.

As for the freeze, no, it does not freeze on the same file.

ealmonte32, 2019-03-01:
> @ealmonte32 - This is running through the Balena cloud. However, I did do an update yesterday to the latest Balena release. I don't appear to be able to run the command you suggested in the terminal window through Balena.
> 
> As for the freeze, no, it does not freeze on the same file.

Hmm.. I haven't started using balena yet so I wouldn't even know why, but when you click on Terminal > Start terminal session.. what do you get? and what happens when you try to run the command? can you post screenshot?

vpetersson, 2019-03-04:
@ealmonte32 - You don't run such commands in Balena as that's all done as part of the Docker build. I suggest you take it for a spin as it's pretty interesting.

ealmonte32, 2019-03-04:
> @ealmonte32 - You don't run such commands in Balena as that's all done as part of the Docker build. I suggest you take it for a spin as it's pretty interesting.

@vpetersson 
So the docker build has its own packages/libraries? I took a quick look at the documentation regarding this, and it says depending on how the app was built, it can share OS libraries or it cannot, depending how the build was made.

Sorry, reason I haven't played around with Balena or learned more about docker is because I don't have extra hardware to practice with.

If the docker is sharing OS libraries, he has to run the commands on the Pi itself then because it seems the packages/libraries are outdated, and just to rule that out they should be updated.. wouldn't that be suggested?

vpetersson, 2019-03-05:
@ealmonte32 

> So the docker build has its own packages/libraries? I 

The bill of material so to speak, is created by the Dockerfile. Hence if you have `RUN apt-get install foobar` in the Dockerfile, whatever version of "foobar" that was available at that point is what will be part of the Docker image. Generally speaking, it does not share libraries with the host system. You can run a Fedora Docker image on an Ubuntu host for instrance. You can find some more info [here](https://docs.docker.com/get-started/).

> If the docker is sharing OS libraries, he has to run the commands on the Pi itself then because it seems the packages/libraries are outdated, and just to rule that out they should be updated.. wouldn't that be suggested?

I thought these were automatically built whenever someone pushed something to the branch. After looking into this however, I realized this is not the case, because Docker Hub does not do ARM builds. We do however have a Raspberry Pi build-farm that we can use.

@rusko124 We need to re-build or Docker images (both for Experimental and for the docker branch in general). Could you create a script to do this? We then need to periodically build these images on our Raspberry Pi build/test farm.

trustees629, 2019-03-06:
Just another follow up. I started a new application in Balena and moved my 2 Pi's over to that new one. I used the same SD cards... Just reflashed them to the new application. They've been running fine for 4 days now. Must've been something in my other Balena application that got corrupted somehow. All seems to be good now.

ealmonte32, 2019-03-06:
> Just another follow up. I started a new application in Balena and moved my 2 Pi's over to that new one. I used the same SD cards... Just reflashed them to the new application. They've been running [...]

Yeah, i really think it was something with outdated packages..

closed 2019-03-07 by vpetersson