Screenly/Anthias #1038: Timezone in experimental

mboehm21, 2019-02-26:
#### Overview of the Issue

Timezone is incorrect in the Docker-Containers. I think you might need to mount /etc/timezone into the containers to use the same settings as the host.

### Environment

* Raspberry Pi Hardware Version: 3
* Screenly OSE Version: newest experimental

vpetersson, 2019-02-26:
@mboehm21 Timezone should not be configured inside the container - that's a system level functionality that should be inherited into the container. Do you have exact steps to reproduce the issue?

mboehm21, 2019-02-28:
I cannot reproduce it now as I don't have access to the screens. But if I remember correctly pages like http://tagesschau.de/infoscreen showed the wrong time. Can you confirm that?

Also "docker exec -it <container> date" showed the wrong time iirc.

vpetersson, 2019-02-28:
I can't reproduce this:

```
$ date
Thu Feb 28 09:44:17 UTC 2019
$ docker run --rm -ti ubuntu date
Thu Feb 28 09:44:20 UTC 2019
```

mboehm21, 2019-03-01:
Well, try to use something else than UTC outside of the container... UTC is default inside of course.

vpetersson, 2019-03-01:
Interesting. I've used Docker for years but never run into this. Perhaps that is because in the server world, UTC is considered best practice. 

```
$ date
Fri Mar  1 09:32:33 PST 2019

$ docker run --rm -ti ubuntu date
Fri Mar  1 17:32:40 UTC 2019
```

Volume mounting is of course an option, but I've frankly never seen that before. Let me think about this for a bit.

vpetersson, 2019-03-04:
Interesting, I just noticed that in [Home Assistant](https://www.home-assistant.io/docs/installation/docker/), they indeed are volume mounting localtime.

reticent98, 2019-03-07:
This is a very minor issue, but it does fit the discussion. I have been testing meteoTV for displaying weather information: https://www.meteoblue.com/en/meteotv/4c0ef9?

With the 11/23/18 image, the time in the bottom corner displays in my timezone. Under with the new experimental build, the time always displays UTC. I'm going to test some other screen-formatted websites and see if they pull the time the same way.

vpetersson, 2019-03-07:
@reticent98 This is expected as per the above issue.

agalewsky, 2019-03-07:
I can confirm as well with experimental having only UTC is problem.   We have several screens with clocks and it is problematic to say the least.   We are trying to use experimental to see if it fixes the SSL hang problem.

PatrickRUG, 2019-03-12:
I also having issues with this with clocks, our pi is configured at CET, but the docker image is in UTC, therefore DST will not be detected by our clock page.

vpetersson, 2019-03-12:
@rusko124 could you fix this - it's a trivial fix it seems like.

bcherup, 2019-03-21:
Any way for a quick patch ?

bcherup, 2019-03-22:
The fix did not work as Ive already tried this. Heres the command you should run.

pi@raspberrypi:~ $ docker exec -it screenly-ose-viewer date
Fri Mar 22 15:40:21 UTC 2019
pi@raspberrypi:~ $

From what I've read the timezone setting needs passed on the startup line of the container.

bcherup, 2019-03-22:
For the record I rebooted after I updated the file. But I was able to fix the issue with 

git reset --hard
git pull

and restart !

vpetersson, 2019-03-22:
Awesome. Closing out.

closed 2019-03-22 by vpetersson