Screenly/Anthias #1041: Website does not render (SonarQube)  Sprint 8 and Experimental

lord-carlos, 2019-03-04:
#### Overview of the Issue

I tried both latest release and experimental. Both version just show the white background instead of the expected webpage.
The webpage is a local version of sonarqube, here is a public one that also shows the same behavior: https://next.sonarqube.com/sonarqube/projects?sort=-analysis_date

#### Reproduction Steps

* Press Add Asset
* Enter the Asset URL https://next.sonarqube.com/sonarqube/projects?sort=-analysis_date
* Press "On"
* A spinner shows and then I just see white

### Environment

* Raspberry Pi Hardware Version: I think it's 3 or 3+
* Screenly OSE Version: Sprint 8 and Experimental.

vpetersson, 2019-03-04:
I don't think there is much we can do on our end to solve this, as that is a limitation in the browser.

lord-carlos, 2019-03-04:
Ok. Thanks anyway.

closed 2019-03-04 by lord-carlos