Screenly/Anthias #1042: 100% CPU from uzbl-core

vern3851, 2019-03-06:
Hi
We've been running screenly OSE for a few years now pointed at web-published google slide.
This has allowed front of office staff the ability to control whats on our notice boards by simply editing a google slide.
Screenly OSE plays the slide for 60min and then a 2sec JPG so that the google slide is re-loaded hourly.

Recently they have added an animated slide. Whenever the slide with the animation (basic gif) plays uzbl-core hits 100% CPU.
The symptoms from noticeboard are the slide before the animated slide plays for an extended time and then the animated slide only plays for half a second.

Latest everything by means of:
`sudo apt-get update -y && sudo apt-get upgrade -y`
`sudo apt-get dist-upgrade`
`bash <(curl -sL https://www.screenly.io/install-ose.sh)`
Cheers
[image]

vpetersson, 2019-03-07:
> Recently they have added an animated slide. Whenever the slide with the animation (basic gif) plays uzbl-core hits 100% CPU.

This is not all that unexpected. `uzbl` does not support hardware accelleration. Hence everything is running on the CPU (and not GPU). 

This is why we began the work on the experimental browser that does indeed support GPU acceleration. Hence I suggest you take it for a spin to see if that solves the problem.

vern3851, 2019-03-07:
sadly no improvement, its just `ScreenlyWebview` thats blasting the CPU this time.
Also noticed that after upgrading to experimental i cant get to the console now via `CTRL + ALT + F1`, had to add an ssh file to the root and ssh in.

ealmonte32, 2019-03-13:
@vern3851 
Have you played around with giving negative nice values to user pi ?
Can you share the animated gif so I can reproduce or is it too private?

vern3851, 2019-03-20:
Sorry not sure what negative nice values means.
However due to time constraint ive moved to another vendor for now.

closed 2019-03-20 by vern3851