Screenly/Anthias #1043: No database migration on Balena upgrade

axlroden, 2019-03-07:
#### Overview of the Issue

I did a pull and push to my balena environment, and there were new fields in the database. 
The migrate.py script is not run and is not pointing at /data/.screenly but /home/pi/...

#### Reproduction Steps

Steps to reproduce this issue, eg:

* git pull
* git push balena master

### Environment

* Raspberry Pi Hardware Version: pi3
* Screenly OSE Version: from 47a0adb to f16d953

I had to manually run the migration script with an edited line:
Change:
`configdir = os.path.join(os.getenv('HOME'), '.screenly/')`
To:
`configdir = os.path.join(os.getenv('/data/.screenly/')`

vpetersson, 2019-03-07:
@axlroden Hmm no that isn't correct. `os.path.join(os.getenv('/data/.screenly/')` hard codes the path. I know that the Balena guys made some changes to this when they introduced the multi-container system. However, we need to make it dynamic such that it works both with and without Balena. The above change would break anything other than Balena.

axlroden, 2019-03-12:
Ignore what I did to fix the issue, the issue as I see it is there is no migration run on a container update.

vpetersson, 2019-03-18:
@axlroden Yes that is a valid bug.

closed 2019-09-10 by RemingtonA