Screenly/Anthias #1048: Screenly hanging opening mjpg steam

hostheimer, 2019-03-10:
#### Overview of the Issue

System stops with moving throug the assets.
System starts stream but will not continue to go to next asset after defined period and keep hanging in the mjpg stream. Steam also not centerred in the middle of the screen.

#### Reproduction Steps

Steps to reproduce this issue, eg:

* Press Add Asset
* Enter the Asset URL http://88.26.233.51:8080/mjpg/video.mjpg
* Press "On"
* Wait for the asset to show up on the screen

### Environment

* Raspberry Pi Hardware Version: 3a
* Screenly OSE Version: 2018-11-23-Screenly-OSE-lite

vpetersson, 2019-03-11:
Thanks @hostheimer.

rusko124, 2019-03-22:
@vpetersson @hostheimer 
This problem shouldn't be if you upgrade to development version.
Resolved after #1039 PR.

rusko124, 2019-04-05:
@vpetersson  This issue is not reproduced in the development branch, I think we can close it.

vpetersson, 2019-04-05:
Closing out issue. Comment if you can still reproduce it on the development branch, @hostheimer.

closed 2019-04-05 by vpetersson