Screenly/Anthias #1049: Show MAC address on System Info screen

stuporglue, 2019-03-12:
Could the device's MAC address be shown on the System Info page? 

### Use Case

My school's wifi has a captive portal the first time a new device connects. You log in, and then the device's MAC is connected to your school account so they know who to contact when malware is spreading or something nefarious is happening. 

Alternatively, you can provide the device's MAC address directly to the office of IT, and then your device can connect without the captive portal. This is the route I'm wanting to take.

It would be convenient for this sort of use case if the device's MAC address were shown on the System Info page. I know how to connect a keyboard and all that, this is purely a convenience feature request.

willworkforicecream, 2019-03-13:
For clarity, are we talking about the System Info page or the splash page? Maybe it is just my org, but I feel like most places running a captive portal isolate unregistered MACs so I'm not sure that you'd be able to get to the System Info page

stuporglue, 2019-03-13:
I was talking about the System Info page, but I think you're right, it'd need to be on the splash screen. 

Sorry, I'm new to Screenly!

ealmonte32, 2019-03-13:
You talking about this page?

[image]

stuporglue, 2019-03-13:
Yes, that's the one I meant originally. 

As @willworkforicecream indicated though, putting it on the splash screen, or even on the Wifi config screen might be better.

ealmonte32, 2019-03-14:
> As @willworkforicecream indicated though, putting it on the splash screen, or even on the Wifi config screen might be better.

Why would it be better? what does it accomplish? it doesn't show any network info.. just trying to understand what you mean..

I thought you would want to display the ethernet address on the same page as the wifi config one since that comes up automatically when no network has been identified..

stuporglue, 2019-03-14:
> what does it accomplish?

My school has a captive portal, so the WiFi has no password, but you must log in to a webpage using your school account username and password. 

Alternatively you can provide the MAC address of a device to the IT department and then that device will be authorized and you will not need to log in. 

If during the initial WiFi setup process I were able to see the MAC address it would be convenient. Up until needing the MAC address I hadn't needed a keyboard for the Screenly setup. 

> I thought you would want to display the ethernet address on the same page as the wifi config one since that comes up automatically when no network has been identified

I might be using the wrong names for screens. I've used Screenly for a total of about 3 hours now, so if I've mis-spoken, I'm sorry. 

What I'm wanting is a way to see the MAC address at some time during the setup process so that people who need it for captive portals don't have to log into the console or something.

ealmonte32, 2019-03-14:
Gotcha, ok so basically from what I understand, you want the page that looks like this:
https://user-images.githubusercontent.com/13919590/35922228-a09806fc-0be2-11e8-9beb-eabc18ea579d.jpg

https://network23.org/dfs3/files/2016/03/screenly-ip.jpg

(I'm just getting these from google image search just for the example)

You want that page to display the same information that the command below shows?
```
ip address show
```

stuporglue, 2019-03-14:
Yes, that would be great. 

I was able to get Screenly set up, this is just a feature request that might make it slightly easier for people in the same situation I was. 

Thank you!

bcherup, 2019-03-24:
Would something like this work for the start_resin_wifi.py script ? I don't know python at all but it seems easy to implement with a few changes/additions to the resin_wifi and hotspot.html files. 

import netifaces as nif
def mac_for_ip(ip):
    'Returns a list of MACs for interfaces that have given IP, returns None if not found'
    for i in nif.interfaces():
        addrs = nif.ifaddresses(i)
        try:
            if_mac = addrs[nif.AF_LINK][0]['addr']
            if_ip = addrs[nif.AF_INET][0]['addr']
        except IndexError, KeyError: #ignore ifaces that dont have MAC or IP
            if_mac = if_ip = None
        if if_ip == ip:
            return if_mac
    return None

>>> mac_for_ip('###.###.###.###')
'2c:41:38:0a:94:8b'

ealmonte32, 2019-03-27:
@bcherup 
No that is too long, it is as simple as this:
```
>>> import netifaces  (which already gets imported)
>>> addrs = netifaces.ifaddresses('wlan0')  (assuming you want wireless)
>>> addrs[netifaces.AF_LINK]
[{'broadcast': 'ff:ff:ff:ff:ff:ff', 'addr': 'aa:bb:cc:dd:ee:ff'}]  <-- mac address will be shown here
>>>
```

qbicdesign, 2019-04-17:
I'd also like this in the system info page
If you want to set a fixed IP for your Screenlys on your DHCP server, then you need the mac in order to make a reservation. This is basic information that you'd expect to see on any other network device system info page. It's not something that should require jumping through hoops to get.

ealmonte32, 2019-04-17:
@qbicdesign 
See this PR for what rusko is working on with regards to system info page: 
https://github.com/Screenly/screenly-ose/pull/1015
____
Now, about your comment:

> I'd also like this in the system info page
If you want to set a fixed IP for your Screenlys on your DHCP server, then you need the mac in order to make a reservation. This is basic information that you'd expect to see on any other network device system info page. It's not something that should require jumping through hoops to get.

I understand you and I also want this, of course this is basic info and of course it should be included in the system info page, but just remember that this is open source and some things to implement are more important than others, if you have your browser in the Screenly system info page, that means you already have the Pi on your network and you know the IP of it, therefore if you needed to get the MAC to put on DHCP, you simply SSH to it and run command `ip link`. _(ssh is extremely easy to setup prior to even turning on your Pi, by placing empty ssh.txt file in /boot/ of SD card)_

What I am trying to say is, it has not been implemented because there are more important things to implement than this on the system info page, I am not saying it should not be there, but it is actually more important to implement in the page that starts up the resin hotspot page since that will show when the Pi starts up without any network connectivity, dont you agree?

closed 2019-04-27 by vpetersson