Screenly/Anthias #1051: Google Drive connectivity

augisbud, 2019-03-14:
Hello, I'm searching for a way to play files from google drive on a raspberry pi 3b+ running screenly

* Raspberry Pi Hardware Version: RPI3B+
* Screenly OSE Version: latest

vpetersson, 2019-03-18:
@augisbud can you expand a bit more on this use case? What kind of content are you envisioning playing?

qbicdesign, 2019-03-22:
You can play files directly from Google Drive - you have to use the "Publish to web" feature from the file menu in your Google presentation, sheet, document or whatever. I'm using this all the time.

If you're trying to automate a slideshow from photos in Drive, well thats another story. You'd need to use a third party drive add-on for that such as this one: https://chrome.google.com/webstore/detail/photo-slideshow/llgeanpccblagafdmfalehmopdbakhol

augisbud, 2019-03-22:
@vpetersson I want to play video, images, gifs in a drive folder continously, in order by names, not randomly

vpetersson, 2019-03-23:
I see. Well, it's an interesting idea. It is however not something we currently have on the roadmap. We are happy to accept code contributions for this, but I don't think it will make it to our roadmap anytime soon unfortunately.

ealmonte32, 2019-03-25:
@augisbud 
Like @qbicdesign says, using Google Slides and publishing the file, is the best way, add all your pictures and videos, etc, as slides, and make sure that within the presentation settings you set to autoplay the slides, and the videos should autoplay as well.. screenly will load the presentation URL just fine, I've tested this and am currently using it as well.

ealmonte32, 2022-05-19:
@qbicdesign @oakie22 

I recommend to download **raspbian lite buster** directly from Raspberry Pi foundation: http://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2021-05-28/

Burn the image to SD card with balenaEtcher.

Configure the Pi with all normal configurations using `sudo raspi-config` at the terminal (locale, timezone, language, hostname, passwords, memory, etc)

Then run the bash install script.. you MUST select the master development branch so that all of the latest fixes/patches and things we've worked on that have made it function properly for many websites and the way I myself run it.

It is important to note that the Raspbian OS image that might have came with the 2020 or any other Screenly prebuilt image, could have been missing some packages or something that was superseded by some other package that is required once the install script installs some packages that were then newer than what the image came with... long story short, just follow my steps above and dont spend too much time installing the prebuilt image and troubleshooting it.