Screenly/Anthias #1054: Suggestion: Dark(er) splash screen?

madsrh, 2019-03-21:
I've always found the [pure white splash screen](https://camo.githubusercontent.com/5eed8b940f5dba9776024d7c73f992bfd00caa6c/68747470733a2f2f662e636c6f75642e6769746875622e636f6d2f6173736574732f3538313039362f3136323739322f64643534626533342d373763662d313165322d383037392d3938306138326439643730632e706e67) to be kind of eye hurting when booting up Screenly for the first time. This is an issue because Screenly is often used on very large screens. Why not use a "dark-theme" for this?

I haven't found the holy grail regarding the design yet, but here's a (dirty) mockup to illustrate what I mean: 

http://madsrosendahl.dk/screenly/

To be clear:
- Dark doesn't have to be black, just not eyehurting white #ffffff
- I'm not referring to the webinterface (for configuration), ONLY the content that is displayed on the digital sign itself initially.

[image]

rusko124, 2019-03-22:
I really like it!
@vpetersson

L4cko, 2019-03-22:
Hi @madsrh 

Looks ok. Remove the thin white line and make the logo smaller 20%

http://prntscr.com/n1dstw

vpetersson, 2019-03-23:
Thanks @madsrh. Take a look in the master branch (developer version) to see what you think.

closed 2019-03-23 by vpetersson