Screenly/Anthias #1055: Screenly OSE - IP but no ping

pcdoc2010, 2019-03-21:
#### Overview of the Issue

Initial configuration - set to use WIFI.  Reboots with proper WIFI IP, but mgmt console does not connect.  Ping doesn't return any results.

#### Reproduction Steps

Reboot (pull power plug)
Mgmt IP is shown
Will not connect

### Environment

* Raspberry Pi Hardware Version: 3 B+
* Screenly OSE Version: image_2018-11-23-Screenly-OSE-lite.zip

rusko124, 2019-03-22:
Hi @pcdoc2010 !
Are you in the same subnet as the device?

pcdoc2010, 2019-03-22:
Yes.

rusko124, 2019-03-22:
Could you provide `pi@raspberrypi:~ $ ip address` output from this command?
Also, Did you try to connect using another device?

qbicdesign, 2019-03-22:
Worth checking your own firewall settings aren't blocking ICMP ping requests. Also (as in my building) the public wifi network is sandboxed to allow only internet access. All local traffic is blocked. Make sure the wifi network you're connecting to allows local traffic.

ealmonte32, 2019-03-25:
@pcdoc2010 Dan? is that you.. I thought we got you all properly set up on the forums lol

pcdoc2010, 2019-03-25:
Sorry. Yes I am fully working now!
Thanks so much!

ealmonte32, 2019-03-26:
> Sorry. Yes I am fully working now! Thanks so much!
> […](#)
> Sent from my iPhone

When you can, just close this issue, thanks.

closed 2019-03-26 by pcdoc2010

umeju, 2019-03-26:
> @pcdoc2010 Dan? is that you.. I thought we got you all properly set up on the forums lol

Hi
I have the same problem. I'm in a subnet, the pi is shown  192.168.10.137 and i am 192.168.10.77 but i can't see the assets control panel. Any suggestion?
ps
with nmap I can see an android phone on that ip address...(???)

ealmonte32, 2019-03-27:
> Hi
> I have the same problem. I'm in a subnet, the pi is shown 192.168.10.137 and i am 192.168.10.77 but i can't see the assets control panel. Any suggestion?
> ps
> with nmap I can see an android phone [...]

@umeju 
If you are on the same subnet and cannot ping or access the assets dashboard (http://192.168.10.137) , then you need to check your network firewall settings, they might be preventing you from accessing the local subnet.

Can you ping the Pi?

nmap might say android because of classification being wrong.. not sure, but irrelevant if you have access to the Pi and type `ifconfig` and get the IP showing that you just posted, then that is the IP of the Pi.
If you cant reach that IP from another device then that is your network blocking the intra-lan connections, which are usually guest LANs or some isolated walled-garden LAN.. sorry but that is out of our reach..