Screenly/Anthias #1057: Development: Change asset location

qbicdesign, 2019-03-22:
Suggestion

#### Overview of the Issue
Currently if you need to change an asset's URL, the only way to do that is to delete the existing asset and add a new one. 

It would be really nice to be able to edit the URL, as re-adding assets requires re-entering all the date/time, duration and queue settings.

vpetersson, 2019-03-22:
@qbicdesign This is on purpose. There are some checks that runs when you add an asset. If we would allow you to edit an asset it could create a kind of edge case.

ealmonte32, 2019-03-25:
@qbicdesign 
Do you find yourself always setting the date/time for duration to the same "duration" ?
For example, without changing the default behavior for all other screenly users, what would you like the default duration or date/time to be for your personal environment?

vpetersson, 2019-04-27:
Closing out this one. The recommended approach for this if you want to automate it is to delete the asset and add a new one. Tampering with assets can create all sorts of bugs.

closed 2019-04-27 by vpetersson