Screenly/Anthias #1061: Access Screenly from modem with static IP

rdscuba, 2019-03-22:
I have Screenly OSE (Sprint 8) setup on a Raspberry Pi B+.  It's working great.  I have a cable modem with a static IP address from my ISP.  The Raspberry Pi is directly connected to the modem via ethernet, nothing else is or will be connected to it.  

How, do I access Screenly (remotely) - what is the best way, please?

Thank you.

JonnyTech, 2019-03-22:
Have you set up `port forwarding` correctly on your modem?

vpetersson, 2019-03-22:
I would not recommend port forwarding the web interface. The Raspberry Pi does have basic security configurations but it is not (currently) designed to be publicly exposed. If you want to expose it, I recommend setting up a VPN or similar.

Alternatively, Screenly Pro comes with a central management interface.You can find a compare and contrast [here](https://www.screenly.io/screenly-ose-vs-pro/).

closed 2019-03-22 by vpetersson