Screenly/Anthias #1064: Enable overscan functionality

bcherup, 2019-03-29:
Screenly or X11 ignores config.txt overscan options, is there anyway we can add this to the Screenly Environment ? I have a cheap TV that turns on OVERSCAN by default and can only be disabled with the remote.... this setting does not save on input change or tv power cycle 

### Environment

* Raspberry Pi Hardware Version: B+ 3
* Screenly OSE Version: Experimental

vpetersson, 2019-04-01:
@bcherup Screenly doesn't do anything as far as overscan goes. That's all handled by the system. Do however note that videos are played using framebuffer (using `omxplayer`) and web/images are not. Hence, you will have a different experience. What type of assets are you having issues with?

qbicdesign, 2019-04-17:
what model TV is it?

vpetersson, 2019-04-27:
Closing due to lack of response.

closed 2019-04-27 by vpetersson