Screenly/Anthias #1066: Screenly OSE hangs

marcxB, 2019-03-29:
Hi,

The raspberry PI with Screenly OSE is running 24/7 and connected to a TV. We switch off the TV manually by pressing standby. It happened several times now when activating the TV the screen is showing an asset (random) and doesn't continue.

Only powering off and on the raspberry helps solving the issue.

Any thoughts?

* Raspberry Pi Hardware Version: 3B+
* Latest Screenly OSE Version:

bcherup, 2019-03-29:
Are you showing any web assets ? 

I believe they addressed this with the newer browser on experimental

ealmonte32, 2019-04-05:
> It happened several times now when activating the TV the screen is showing an asset (random) and doesn't continue.
> 

You need to provide some more details than that, you can start by pasting here the output of at least the last 20 or so lines of this command:
`journalctl | grep python`

When you turn the tv and see the asset that doesn't continue, do this either via SSH or if physical access to the Pi with keyboard `ctrl+alt+f1`

vpetersson, 2019-04-27:
Closing due to lack of response.

closed 2019-04-27 by vpetersson