Screenly/Anthias #1069: Web asset doesn't load anymore

IsaakGroepT, 2019-04-03:
When filing a bug, please include the following headings if possible. Any example text in this template can be deleted.

#### Overview of the Issue

Since the upgrade to the latest Screenly OSE, our web asset doesn't load anymore. Restarting or reactivating the asset doesn't solve the issue.

I'm seeing this:
Apr 03 09:59:23 raspberrypi python[513]: Asset https://vergaderingen.ambrassade.be/ at https://vergaderingen.ambrassade.be/ is not available, skipping.

#### Reproduction Steps

Steps to reproduce this issue, eg:

* Press Add Asset
* Enter the Asset URL ...
* Press "On"
* Wait for the asset to show up on the screen

### Environment

* Raspberry Pi Hardware Version: v3
* Screenly OSE Version: latest

rusko124, 2019-04-05:
Hi @IsaakGroepT !
`bash <(curl -sL https://www.screenly.io/install-ose.sh)`
Run this script and upgrade to development version. I reproduced issue and this should help you.
Let me know if it helps.

IsaakGroepT, 2019-04-06:
That seems to fix it. Thanks!

closed 2019-04-06 by IsaakGroepT