Screenly/Anthias #1070: Site not showing

Woeka, 2019-04-03:
#### Overview of the Issue

The website is not shown, blank screen. Log shows InsecureRequestWarning. Opening the site directly in uzbl also does not show the site. Updating to development and experimental did not solve the problem.

#### Reproduction Steps
Steps to reproduce this issue, eg:

* Press Add Asset
* Enter the Asset URL https://monitoringpublic.solaredge.com/solaredge-web/p/kiosk?guid=5162e861-c5b9-4807-8821-2fe5ae2a696f&locale=nl_NL
* Press "On"
* Wait for the asset to show up on the screen, only getting partly looks like uzbl hangs with a loading image from the site on the screen.
[screenly_log.txt](https://github.com/Screenly/screenly-ose/files/3039774/screenly_log.txt)

Site in uzbl only shows blank screen, not even loading image.
`uzbl -u https://monitoringpublic.solaredge.com/solaredge-web/p/kiosk?guid=5162e861-c5b9-4807-8821-2fe5ae2a696f -v --dis play=:0.0`
[uzbl_log.txt](https://github.com/Screenly/screenly-ose/files/3039778/uzbl_log.txt)

`uzbl -u https://monitoringpublic.solaredge.com/solaredge-web/p/kiosk?guid=5162e861-c5b9-4807-8821-2fe5ae2a696f --verbose -p --display=:0.0`
[uzbl_verbose_log.txt](https://github.com/Screenly/screenly-ose/files/3039801/uzbl_verbose_log.txt)

Then I tried to update to Experimental branch
```
bash <(curl -sL https://www.screenly.io/install-ose.sh)
updated /etc/systemd/timesyncd.conf
sudo apt update && sudo apt full-upgrade -y
sudo rpi-update
```

This resulted in a TSL handshake problem, after restarting service this did not repeat. 

```
$ python
>>> import requests
>>> requests.head('https://monitoringpublic.solaredge.com/solaredge-web/p/kiosk?guid=5162e861-c5b9-4807-8821-2fe5ae2a696f')
<Response [405]>
```

Modifying /home/pi/screenly/lib/utils.py did remove some errors but did not solve the problem. 

Which web browser is used in the experimental branch? I would love to test the url like uzbl.

Using google sites is not a option, I've tried it but the site is not looking good in google sites And I'm first getting a update browser message and only the bottom bar which invites you to use google sites.

Modifying [viewer.py](https://github.com/Screenly/screenly-ose/issues/1031#issuecomment-466600394) did not work.

#### Environment
* Raspberry Pi Hardware Version: 3 model B v1.2
* Screenly OSE Version: Development Branch / Experimental branch

ealmonte32, 2019-04-05:
@Woeka 
To stop request module from throwing urllib error InsecureRequestWarning, you need to edit the file:
`sudo nano ./screenly/lib/utils.py`

[image]

___
Regarding the solaredge page not loading, I think this is because uzbl doesnt support web components.. but I do remember that someone said it was working for them one day and then not the next, see this issue and comment : https://github.com/Screenly/screenly-ose/issues/1014#issuecomment-460302383

So maybe it is something with a shared library, we would have to see all changes done to libraries/dependencies from the version that user was using to now.. or maybe solaredge updated their backend framework and uses something uzbl doesnt understand/ cant render?

Woeka, 2019-04-08:
Hi, Thanks for your reply. 

I've changed utils.py and there are less errors shown. But it did not solve the problem.

I had it running on a previous version of Screenly from 2016. So like you note SolarEdge probably updated something. I thought I would fix it with updating to the experimental branch with a more recent browser, however it didn't. Which browser is used?

ealmonte32, 2019-04-08:
@Woeka 
> I had it running on a previous version of Screenly from 2016. So like you note SolarEdge probably updated something. I thought I would fix it with updating to the experimental branch with a more recent browser, however it didn't. Which browser is used?

The browser used in production and developer is UZBL, the one used in Experimental as far as I know is Qt Web.

You say you had it running on a 2016 version of screenly?
Just so I understand correctly because specifics here matter, you had a version/image of screenly that was installed/setup in 2016 and until recently (without you making any changes to screenly) it was able to display the Solaredge page properly?
Or did you upgrade screenly because the page no longer started to display and you thought the new version of screenly would work?

Woeka, 2019-04-09:
It stopped working totally also had to replace the Raspberry. 

A saved image of the old setup of 2016 (working at the time) did not show the site at the moment. Hoping a upgrade to the development branch would fix that, but it didn't.

ealmonte32, 2019-04-11:
Ok so if the saved image from 2016 was showing the page in the year 2016 but now recently it stopped showing it without you upgrading or touching screenly version, it could only mean the solaredge page changed their html code..
Only certain browsers can take advantage of new web technology and the current browser on screenly does not seem to support these new html5 web components (a view-source on solaredge tells me it is html5 and using features only html5 compatible browsers can show).

This is a html5 test from uzbl, and as you can see from the html5 test for web components, it doesn't really support anything..

[image]

vpetersson, 2019-04-11:
@ealmonte32 this is why we began the work on the experimental branch with the new browser (which subsequently removes Uzbl). I guess we need to make a final push here to get it out the door.

vpetersson, 2019-04-27:
FYI - I've scheduled the work for the final push to deprecate UZBL in Sprint 12.

closed 2019-07-03 by stale[bot]