Screenly/Anthias #1074: this url doesn't render very well

pooh22, 2019-04-05:
#### Overview of the Issue

At the school where I installed screenly ose sprint 8, this url doesn't display correctly:
https://www.dedagvandaag.nl/agenda/groep678/4/2019

If you open it in a regular browser it shows a perfect page for the TV in the school, but the html renderer on screenly messes it up.

#### Reproduction Steps

Steps to reproduce this issue, eg:

* Press Add Asset
* Enter the Asset URL https://www.dedagvandaag.nl/agenda/groep678/4/2019
* Press "On"
* Wait for the asset to show up on the screen

Or you can call uzbl-core directly:
```
uzbl-core --geometry=1920x1080 https://www.dedagvandaag.nl/agenda/groep678/4/2019
```
(I've tested this with a self compiled checkout of master from the uzbl git repo on an ubuntu 18.04 machine)

### Environment

* Raspberry Pi Hardware Version: rpi 3B+
* Screenly OSE Version: sprint 8

### Expected/hoped for fix

I'd like to know if it's possible to use another renderer or a fix for uzbl for screenly, so this (and other) url will display like in a browser on a desktop.
Possibly this will be fixed in sprint 9, when it's done. That will be fine for me too.

ealmonte32, 2019-04-11:
> (I've tested this with a self compiled checkout of master from the uzbl git repo on an ubuntu 18.04 machine)

Quick question, you replaced the last uzbl version that comes from the debian stretch dist which is version 20120514-1.2+b2 , with this compiled version from the uzbl github? _(https://github.com/uzbl/uzbl)_

And were you or were you not able to get the page to render properly with this compiled version from the uzbl github which is obviously newer by maybe 2 or 3 years?

pooh22, 2019-04-12:
I didn't change the screenly installation, I just tested with the github version of uzbl on my desktop with ubuntu 18. This test also didn't render the page correctly.

My questions are basically:
- can I use another html renderer than uzbl?
- is there another way this page can render correctly on screenly?

vpetersson, 2019-04-12:
@pooh22 Please try switching to the experimental branch. It moves away from UZBL to a more modern browser.

pooh22, 2019-04-12:
OK, I will try when I can spend a bit of time on this...

vpetersson, 2019-04-27:
Closing due to inactivity. @pooh22 please feel free to re-open the issue if you're able to reproduce under the experimental branch.

closed 2019-04-27 by vpetersson

pooh22, 2019-05-05:
I tried experimental today, the page still doesn't render correctly