Screenly/Anthias #1076: 'Unable to resolve IP address' still comes with latest NOOBS

kams01, 2019-04-08:
When filing a bug, please include the following headings if possible. Any example text in this template can be deleted.

#### Overview of the Issue

I recently starting working on this as a user to setup OSE at our facility. I used NOOBS to install OSE. After you move the Pi to another network, the network selection tool does not pop back up after "Unable to resolve IP"

Since then I have used several things - Updated to latest build <- by the way, how do I know which version of OSE am I running? there is no simple way to know this!

Updated to the latest development build (all using SSH or local)
Brought it back to the original network where it attached and hit the "Rerun Network Detection" and it does not reset anything.
Tried to remove the wifi-set file, but it cannot find it.
So basically had to reinstall OSE at the final location and deploy it there - not very practical if you want to test a few things out earlier.

So am I missing anything here?

#### Reproduction Steps

Setup Screenly on one network.
Now try to move it to another network, or disable your WIFI AP for a bit
Pi will boot into Screenly and get stuck on 'Unable to...'

### Environment

* Raspberry Pi Hardware Version: Pi 2, Pi 3B+, Pi 3A
* Screenly OSE Version: How do I find this?!

bartv, 2019-04-09:
We see this issue as well. It worked perfect one network. When we move it to another network we get "Unable to resolve ip address" and the slideshow starts. The initial network setup is not restarted.

ealmonte32, 2019-04-11:
> Since then I have used several things - Updated to latest build <- by the way, how do I know which version of OSE am I running? there is no simple way to know this!

There will be one simple way to know this very soon.
___

I'll let viktor chime in about the wifi reset when moved to another place where the setup SSID is not broadcasting..

I'll try to reproduce your steps exactly, although I have already experienced the part where if assets were uploaded prior to moving to another location, then it starts to cycle through those assets instead of the URLs (since they are not available it skips these), sort of like an 'offline' mode. But this gets resolved when it gets connectivity again.
___
>So basically had to reinstall OSE at the final location and deploy it there - not very practical if you want to test a few things out earlier.

There was no need to reinstall OSE at a final location, you could have simply setup the wifi there using the keyboard and getting into console, a few commands and you would have been connected.
___
My opinion:
I think it should not be that if your Pi doesn't have network connectivity that it automatically starts up the resin wifi connect screen, that means any time the interface is detected as down or without an IP this would try to come up (that means it would need to use the interface for this every time and not allow it to obtain itself an address from dhcp, and how about a timeout? what if the DHCP server or AP just doesn't give the device an IP at that moment, it would constantly have that Ad-hoc/hotspot screen on), that would not be a good solution.
Resetting the wifi config prior to moving the Pi to its location should be one of the many ways to get the Pi connected. We have to focus on that part, specifically why after you clicked "Re-run network detection" you did not get the WiFi setup screen when Pi was rebooted in new location. I'll be testing for this.

vpetersson, 2019-04-17:
> I think it should not be that if your Pi doesn't have network connectivity that it automatically starts up the resin wifi connect screen

The problem with this is that can create another issue. Imagine that you have three OSE boxes up and running and they work just fine with the WiFi. You then need to do some maintenance on the WiFi, which causes the devices to disconnect. If we then automatically launch the WiFi wizard, we a) disconnect them from their previous connection and b) impact the playback of the screen. You now need to manually reconnect all of these devices to the WiFi.

As such, I'm fairly confident that the only way to do this properly is to have this reset being manual. If not, it will cause a lot of unexpected issues.

ealmonte32, 2019-04-17:
> The problem with this is that can create another issue. Imagine that you have three OSE boxes up and running and they work just fine with the WiFi. You then need to do some maintenance on the WiFi, [...]

@vpetersson I think you thought I said the opposite, I am actually saying exactly that, agreeing with what you are saying as your explanation, that **it is not** a good idea to disconnect the wifi just because it loses connectivity once it has already been set up with wifi, meaning I agree that you should have a manual reset for this. Maybe the way I worded the sentence wasn't good, putting a double negative at beginning ;)

kams01, 2019-04-17:
Thanks for the input @ealmonte32 and @vpetersson 

I agree with you on that front, however the key issue is and remains, how does one change the wifi settings:

You say do it manually via the WEBGUI but as I have mentioned, it does not work. None of the options in various threads worked for me to reset the network settings while in terminal mode either. So simple question is, when I move the device from one network to another, how do I change the network config?

so far, Ive only been able to do this by reinstalling OSE through NOOBS. I have also tried the experimental and development versions but the reset methods do not work. Thanks for your comments

ealmonte32, 2019-04-17:
@kams01 
Just to confirm, you are saying that when you click `Re-run network detection` nothing happens? After clicking that and rebooting the Pi, you dont get the Resin WiFi Connect screen?

kams01, 2019-04-17:
That is correct. When I click Re-run network detection, nothing happens after reboot. the Pi just connects back to the network and displays the IP to access its web gui

vpetersson, 2019-04-23:
@kams01 and you're running on the latest developer version?

kams01, 2019-04-23:
The issue was seen on the latest version also once I updated it via terminal (again I have been asking how I can find out the SW Version # but so far no one has told me) however I only had 1 or 2 slides.

vpetersson, 2019-04-27:
@kams01 - new release is coming today to NOOBS (hopefully). Please re-open the issue if it still is reproduceable.

> I have been asking how I can find out the SW Version # but so far no one has told me

We use rolling releases, so we don't have [traditional semantic versioning](https://semver.org/), but it's something we might adopt in the future.

closed 2019-04-27 by vpetersson

ealmonte32, 2019-04-28:
Remember we added the temp simple way of seeing which version of screenly user selected with file `~/OSE_version.md`

vpetersson, 2019-04-29:
Or as I discovered this weekend, in the system info page. :)