Screenly/Anthias #1082: Asset address clipped

qbicdesign, 2019-04-17:
#### Overview of the Issue

After editing a URL asset, the asset URL is shown clipped if the URL is long

#### Reproduction Steps

* Press Add Asset
* Enter a long Asset URL such as a published to web Google slide
* Save it
* Go back to edit the asset
* You'll see that the URL is clipped, and attempting to copy/paste the URL into browser for checking purposes will result in error

### Environment

* Raspberry Pi Hardware Version: Model 3B+ Revision: 1.3 
* Screenly OSE Version: development@79b0818 and possibly earlier

This was not a problem in the 2018-05-04 custom image

vpetersson, 2019-04-17:
@qbicdesign sure - but I don't understand why this is a problem?

qbicdesign, 2019-04-18:
if you're trying to check the url is correct, troubleshooting non-display etc

vpetersson, 2019-04-18:
I'd prefer to keep it truncaded as it looks a lot better for long URLs. However, we could add something like a "copy to clipboard" feature perhaps?

JonnyTech, 2019-04-18:
If this also applied to filenames, can you truncate the central part of the filename? Most of my images are identifiable by the last characters, ie `very long filename of picture 01.jpg`, `very long filename of picture 02.jpg`, etc. Seeing the end of the filename would be preferable. This may also be the case with web / url assets.

vpetersson, 2019-04-21:
@JonnyTech sure, but isn't a better solution to rename the asset such that it has sensible name?

JonnyTech, 2019-04-21:
Yes, indeed, if it were me. But at our church the older lay folks struggle with such simple tasks. No matter how simple I make the instructions for them. The aim should be to make it easy and accessible for all to manage their assets. But thanks @vpetersson for replying, it is not a Screenly problem, rather a user / [PICNIC](https://en.m.wikipedia.org/wiki/PICNIC) issue

vpetersson, 2019-04-27:
Looked quickly into this. You can get the URL by pressing 'Edit' and on that page you can see the full asset URL.

closed 2019-04-27 by vpetersson