Screenly/Anthias #1083: Help Please - trying to push Screenly-ose to raspberry pi3 via Balena cloud

Dereshp, 2019-04-18:
Hi all, absolutely new to this and hoping for some help
#### Overview of the Issue
I have loaded Balena cloud and created my device herein.  I am struggling to push content from git/local to the device

A paragraph or two about the issue you're experiencing.
I keep receiving the below error and have not being successful in moving any data across to th pi
fatal: unable to access 'https://github.com/Screenly/screenly-ose.git/': The requested URL returned error: 403

#### Reproduction Steps
Username for 'https://github.com': dereshpillay@icloud.com
Password for 'https://dereshpillay@icloud.com@github.com': 
remote: Permission to Screenly/screenly-ose.git denied to Dereshp.
fatal: unable to access 'https://github.com/Screenly/screenly-ose.git/': The requested URL returned error: 403
Dereshs-MacBook-Pro:screenly-ose dereshpillay$ git push
Username for 'https://github.com': dereshp
Password for 'https://dereshp@github.com': 
remote: Permission to Screenly/screenly-ose.git denied to Dereshp.
fatal: unable to access 'https://github.com/Screenly/screenly-ose.git/': The requested URL returned error: 403
Dereshs-MacBook-Pro:screenly-ose dereshpillay$ git push
Username for 'https://github.com': Dereshp
Password for 'https://Dereshp@github.com': 
remote: Invalid username or password.
fatal: Authentication failed for 'https://github.com/Screenly/screenly-ose.git/'
Dereshs-MacBook-Pro:screenly-ose dereshpillay$ git push
Username for 'https://github.com': Dereshp
Password for 'https://Dereshp@github.com': 
remote: Permission to Screenly/screenly-ose.git denied to Dereshp.
fatal: unable to access 'https://github.com/Screenly/screenly-ose.git/': The requested URL returned error: 403
Dereshs-MacBook-Pro:screenly-ose dereshpillay$ 

### Environment
MacOS 
* Raspberry Pi Hardware Version:3
* Screenly OSE Version:

erykosky, 2019-04-19:
I can't push it too. 

[main]     Step 1/27 : FROM resin/raspberrypi3-64-debian
[main]     repository resin/raspberrypi3-64-debian not found: does not exist or no pull access

vpetersson, 2019-04-20:
@Dereshp That sounds like a Balena usage issue - please post on their [forum](https://forums.balena.io)

@erykosky This is a duplicate of #1080. We essentially need to move to a different image given some recent changes on their end.

closed 2019-04-20 by vpetersson