Screenly/Anthias #1088: Consistant rebooting on specific web pages

mtj2k, 2019-04-25:
When filing a bug, please include the following headings if possible. Any example text in this template can be deleted.

#### Overview of the Issue

My Screenly crashes on specific webpages consistantly. They are pages which used to work OK on my Pi 2 and now fail on 3B+. I am unable to view the logs since it keeps rebooting when the error occurs.

#### Reproduction Steps

Steps to reproduce this issue:

* Press Add Asset
* Enter the Asset URL http://www.biltonsilver.co.uk/
* Press "On"
* Wait for the asset to show up on the screen
* Screenly crashes and returns to IP address screen

### Environment

* Raspberry Pi Hardware Version: Model 3B+
* Screenly OSE Version: 2018-11-23

vpetersson, 2019-04-27:
It's possible that this is due to the soundcloud extension. Please try the experimental branch and see if that resolves the issue.

RemingtonA, 2019-06-30:
Can no longer reproduce on Screenly OSE (version 2019-04-27).

closed 2019-06-30 by vpetersson