Screenly/Anthias #1091: Upgrade to Python 3

vpetersson, 2019-04-27:
When we created Screenly OSE, Python 3 was still in its early days. Things have changed, and Python 2.7 is reaching EOL next year. It's time to start upgrading our code base to Python 3.

ekevoo, 2019-06-27:
Hello, I have ported a different project from 2 to 3 in the past and I had zero problems, though it was definitely much smaller. In #1045 you mention that it's partly due to dependencies. Have you already identified which dependencies are affected?

vpetersson, 2019-06-28:
@ekevoo We'd love help with the upgrade to Python 3. Things have likely changed sin the initial comment in 1045, so I don't know if it is still the case. My gut feeling tells me that by now everything will work, and we will just need to refactor the code.

ekevoo, 2019-06-28:
I might be able to help, but I just installed Screenly for the first time yesterday and haven't looked at the code yet. I guess the two big initial questions are

1. What's the situation with unicode strings and byte strings? Are they marked (u"" and b"" or are there usually `from __future__ import unicode_literals`)?
2. I notice the installer spends quite some time and effort clearing out any potential `python3` and manually pulls `pip`, which is what made me come look for this issue; is there an easy way to disable that? I don't want to have to wrestle with it for days.

vpetersson, 2019-06-28:
> What's the situation with unicode strings and byte strings? Are they marked (u"" and b"" or are there usually from __future__ import unicode_literals)?

That's a good question - @rusko124?

> I notice the installer spends quite some time and effort clearing out any potential python3 and manually pulls pip, which is what made me come look for this issue; is there an easy way to disable that? I don't want to have to wrestle with it for days.

Sure, that's done in [here](https://github.com/Screenly/screenly-ose/blob/master/bin/install.sh).

ekevoo, 2019-07-07:
So I'm working at this today and I'm kind of puzzled on amf.py. I'm not sure what this does but it's clearly heavily encoding-related. I saw very few references to py2's str which is equivalent to py3's bytes.

vpetersson, 2019-07-08:
@rusko124 could you chime in here?

ekevoo, 2022-04-11:
@vpetersson Hey are you still interested in my help? My previous pull request is super stale and no longer usable, but I would appreciate if we could have a 30~60min meeting just to help me set up a testing environment so I can start over and build the PR in a way that the CI tools will be appeased.

vpetersson, 2022-09-23:
@ekevoo as you might have seen, we've resurrected this project a bit lately. An engineer is about to start working on OSE and one of the first tasks will be to start working on the Python 3 migration. If you're still interested in helping out, please let us know.

ekevoo, 2022-09-27:
Sure, no problem. Can I send you a calendar link privately? My Gmail username is the same as here.

vpetersson, 2022-09-27:
@ekevoo I've invited you to our Slack channel for Screenly OSE.

PaoloSymple, 2022-10-14:
Hello guys im trying to upgradeto python 3.9 on raspberry legacy 32 bit on pi 8gb but unfortunately i cant  the reason its when i attempt to install screenly i receive this "could not find a version that satisfies the requirement ansible-core==2.12 " i read about it and in my understanding i must run python 3.9 , does any one faced that issue. Again its a Raspberian Legacy 32 bit /not lite version/

vpetersson, 2022-10-14:
First, *never* use the Desktop version of Raspbian/Raspberry Pi OS with Screenly. It will not work.

Second, the error your facing sounds like you're still on the `production` branch. Switch over to `master` and this should be solved.

PaoloSymple, 2022-10-14:
ok so somehow i was able to start the process under raspberian legacy but
therefore i facing an error for could not find a version that satisfies the
requirement ansible-core==2.12
and im looking in the link u  send me there is nothing for that , when i
tried to run it in lite version i cant get nowhere

vpetersson, 2022-10-14:
Start over afresh and follow [these instructions](https://github.com/screenly/screenly-ose#installing-on-raspbianraspberry-pi-os).

PaoloSymple, 2022-10-14:
ok ill give it a shot just to confirm u want me to install the raspberry pi os legacy lite ?

vpetersson, 2022-10-14:
No need for the legacy version. The latest (Bullseye Lite) should work just fine.

PaoloSymple, 2022-10-14:
bash :/dev/fd/631: No such file or directory its the issue when i run it the way you said

vpetersson, 2022-10-14:
Start with a new SD card and re-flash it from the start. Your installation or SD card is likely broken.

PaoloSymple, 2022-10-14:
the SD card that i have its brand new actually from yesterday old

PaoloSymple, 2022-10-14:
now im getting Sceenly OSE must be installed as the user pi with sudo premission when i type sudo i get the exat same error i gave it to ya ..

PaoloSymple, 2022-10-14:
[image]
 this is what it comes when I run the script

vpetersson, 2022-10-17:
Based on the information, you're running Raspbian/Raspberry Pi OS **Buster**, not **Bullseye**. We haven't tested this on Buster.

nicomiguelino, 2023-03-03:
#1743 was already merged. Closing this ticket now...
Thanks, @vpetersson and @EliseAv for your help, especially for the references.

closed 2023-03-03 by nicomiguelino