Screenly/Anthias #1094: Fix dependency issue

vpetersson, 2019-04-27:
When creating the latest disk image, I ran across the following error:

```
ERROR: requests 2.20.0 has requirement urllib3<1.25,>=1.21.1, but you'll have urllib3 1.25.1 which is incompatible.
ERROR: cryptography 2.6.1 has requirement cffi!=1.11.3,>=1.8, but you'll have cffi 1.7.0 which is incompatible.
```

We should address this going forward.

gstoelen, 2019-05-02:
During installation of the latest build I also ran into this error : 
/home/pi/.local/lib/python2.7/site-packages/requests/__init__.py:91: RequestsDependencyWarning: urllib3 (1.25.1) or chardet (3.0.4) doesn't match a supported version!
  RequestsDependencyWarning)

It's refusing to start the Screenly services. 

This comes from the logs : 

May 02 13:45:21 ZNCTPI001 systemd[1]: screenly-viewer.service: Service hold-off time over, scheduling restart.
May 02 13:45:21 ZNCTPI001 systemd[1]: Stopped Screenly Viewer.
May 02 13:45:21 ZNCTPI001 systemd[1]: Starting Screenly Viewer...
May 02 13:45:21 ZNCTPI001 systemd[1]: Started Screenly Viewer.
May 02 13:45:24 ZNCTPI001 python[2248]: /usr/local/lib/python2.7/dist-packages/requests/__init__.py:91: RequestsDependencyWarning: urllib3 (1.25.1) or chardet (3.0.4) doesn't match a supported version!
May 02 13:45:24 ZNCTPI001 python[2248]:   RequestsDependencyWarning)
May 02 13:45:26 ZNCTPI001 python[2248]: Viewer crashed.
May 02 13:45:26 ZNCTPI001 python[2248]: Traceback (most recent call last):
May 02 13:45:26 ZNCTPI001 python[2248]:   File "/home/pi/screenly/viewer.py", line 516, in <module>
May 02 13:45:26 ZNCTPI001 python[2248]:     main()
May 02 13:45:26 ZNCTPI001 python[2248]:   File "/home/pi/screenly/viewer.py", line 481, in main
May 02 13:45:26 ZNCTPI001 python[2248]:     wireless_connections = nmcli_get_connections('wlan*', 'ScreenlyOSE-*', active=True)
May 02 13:45:26 ZNCTPI001 python[2248]:   File "/home/pi/screenly/lib/utils.py", line 123, in nmcli_get_connections
May 02 13:45:26 ZNCTPI001 python[2248]:     out = sh.nmcli('-t', '-f', ','.join(fields), 'c', 'show', *params)
May 02 13:45:26 ZNCTPI001 python[2248]:   File "/usr/local/lib/python2.7/dist-packages/sh.py", line 1654, in __getattr__
May 02 13:45:26 ZNCTPI001 python[2248]:     return self.env[name]
May 02 13:45:26 ZNCTPI001 python[2248]:   File "/usr/local/lib/python2.7/dist-packages/sh.py", line 1593, in __getitem__
May 02 13:45:26 ZNCTPI001 python[2248]:     return Command._create(k, **self.baked_args)
May 02 13:45:26 ZNCTPI001 python[2248]:   File "/usr/local/lib/python2.7/dist-packages/sh.py", line 498, in _create
May 02 13:45:26 ZNCTPI001 python[2248]:     if not path: raise CommandNotFound(program)
May 02 13:45:26 ZNCTPI001 python[2248]: CommandNotFound: nmcli
May 02 13:45:26 ZNCTPI001 python[2248]: Traceback (most recent call last):
May 02 13:45:26 ZNCTPI001 python[2248]:   File "/home/pi/screenly/viewer.py", line 516, in <module>
May 02 13:45:26 ZNCTPI001 python[2248]:     main()
May 02 13:45:26 ZNCTPI001 python[2248]:   File "/home/pi/screenly/viewer.py", line 481, in main
May 02 13:45:26 ZNCTPI001 python[2248]:     wireless_connections = nmcli_get_connections('wlan*', 'ScreenlyOSE-*', active=True)
May 02 13:45:26 ZNCTPI001 python[2248]:   File "/home/pi/screenly/lib/utils.py", line 123, in nmcli_get_connections
May 02 13:45:26 ZNCTPI001 python[2248]:     out = sh.nmcli('-t', '-f', ','.join(fields), 'c', 'show', *params)
May 02 13:45:26 ZNCTPI001 python[2248]:   File "/usr/local/lib/python2.7/dist-packages/sh.py", line 1654, in __getattr__
May 02 13:45:26 ZNCTPI001 python[2248]:     return self.env[name]
May 02 13:45:26 ZNCTPI001 python[2248]:   File "/usr/local/lib/python2.7/dist-packages/sh.py", line 1593, in __getitem__
May 02 13:45:26 ZNCTPI001 python[2248]:     return Command._create(k, **self.baked_args)
May 02 13:45:26 ZNCTPI001 python[2248]:   File "/usr/local/lib/python2.7/dist-packages/sh.py", line 498, in _create
May 02 13:45:26 ZNCTPI001 python[2248]:     if not path: raise CommandNotFound(program)
May 02 13:45:26 ZNCTPI001 python[2248]: sh.CommandNotFound: nmcli
May 02 13:45:26 ZNCTPI001 systemd[1]: screenly-viewer.service: Main process exited, code=exited, status=1/FAILURE
May 02 13:45:26 ZNCTPI001 systemd[1]: screenly-viewer.service: Unit entered failed state.
May 02 13:45:26 ZNCTPI001 systemd[1]: screenly-viewer.service: Failed with result 'exit-code'.
May 02 13:45:26 ZNCTPI001 systemd[1]: screenly-viewer.service: Service hold-off time over, scheduling restart.
May 02 13:45:26 ZNCTPI001 systemd[1]: Stopped Screenly Viewer.
May 02 13:45:26 ZNCTPI001 systemd[1]: Starting Screenly Viewer...
May 02 13:45:26 ZNCTPI001 systemd[1]: Started Screenly Viewer.
May 02 13:45:30 ZNCTPI001 python[2268]: /usr/local/lib/python2.7/dist-packages/requests/__init__.py:91: RequestsDependencyWarning: urllib3 (1.25.1) or chardet (3.0.4) doesn't match a supported version!
May 02 13:45:30 ZNCTPI001 python[2268]:   RequestsDependencyWarning)
lines 1997-2051/2051 (END)

rusko124, 2019-05-02:
@gstoelen It looks like you don't have network-manager installed. Also when you are updated I recommend to agree with this `Do you want Screenly to manage your network? This is recommended for most users. (Y/n)`

gstoelen, 2019-05-02:
@rusko124 Installing network-manager immediately solved it :)  Thx!

rusko124, 2019-05-02:
@gstoelen I was glad to help!

@vpetersson Could you help me, please? In the last image, we are setting up the network, so this does't concern them. Some users as I see, `network-manager` is not installed. But we use `nmcli` to configure and reset the network in the latest updates. Should we handle the `sh.CommandNotFound` error? or we can install the `network-manager` in `ansible` for example.

vpetersson, 2019-05-03:
@rusko124 yeah sure we can include that in ansible but also take a look at my comment in #1101.