Screenly/Anthias #1097: Perform facelift of graphic assets

vpetersson, 2019-05-01:
The assets are becoming somewhat dated. We need to:

 * Update the splash screen to make it clear that it's Screenly OSE (i.e perhaps write it out "Open Source Edition" 
 * Refresh all the look and feel to to make it consistent across all assets (i.e. not mix the blue and purple color scheme)

vpetersson, 2019-05-08:
ping @L4cko

L4cko, 2019-05-21:
@vpetersson Here is the updated kit https://drive.google.com/open?id=1xgg_5QTA8hcZUwLh5Hz5DFxRw9WxBOS4

vpetersson, 2019-05-22:
Thanks @L4cko. I don't see the exact assets requested above in the folder, but please collaborate with @rusko124 on getting this implemented.

rusko124, 2019-05-23:
@L4cko It would be great if we could also provide a design for this #1107

L4cko, 2019-05-27:
[image]

This is the new footer design that I and Alexander work on.

L4cko, 2019-05-27:
[image]
This is the new footer

vpetersson, 2019-05-27:
@L4cko No it isn't. That's the footer for the marketing site (screenly.io). We're talking about the footer in OSE here.

L4cko, 2019-05-28:
[image]
[image]

Here are two versions. If you are updating the blue to purple, let me know.

rusko124, 2019-05-28:
@L4cko Look good to me.
As for the colours I think need a full design to discussing.
#842 there @madsrh already provided some design too.

vpetersson, 2019-06-17:
@rusko124 let's go with this one:
[image]

closed 2019-08-23 by stale[bot]