Screenly/Anthias #1098: Screenly does not provide a method to input username in initial Wifi setup

djflux, 2019-05-01:
#### Overview of the Issue

Stock Screenly OSE does not provide a method to input a username to connect to a WPA2 SSID.

[image]

#### Reproduction Steps

* Load Pi with 2019-04-27-Screenly-OSE-lite.img
* Boot Pi
* Connect to adhoc SSID
* Navigate to http://screenly.io/

### Environment

* Raspberry Pi Hardware Version: Pi3 Rev B from adafruit 
* Screenly OSE Version: 2019-04-27-Screenly-OSE-lite.img

vpetersson, 2019-05-02:
@djflux Assuming you're referring to WPA2 Enterprise (i.e. ~802.1x), this is far more complex than just adding a field for username and password. It's also not supported in Belena/Resin wifi connection tool. 

You can still configure the Rasperry Pi to connect to this manually, but this is out of scope for our use case for the WiFi connection tool unfortunately. If you want, please feel free to lobby for this feature [here](https://github.com/balena-io/wifi-connect). If it is added there, we will inherit it.

djflux, 2019-05-02:
Thanks for the quick response. Searching through the WiFi Connect issues, according to https://github.com/balena-io/wifi-connect/issues/123 it looks as if the capability to support WPA Enterprise EAP-PEAP (which is exactly what I need) was added to wifi-connect v4.2.0 with this commit: 

https://github.com/balena-io/wifi-connect/commit/6547a4a77c76d0ec903033f44a177f312c2831fe

Is there something I need to do on my end to get it working in the UI. It's be willing to do the work and submit a pull request if someone points me in the correct direction :)

Thanks for the great software!

vpetersson, 2019-05-03:
@rusko124 let's upgrade WiFi Connect.

vpetersson, 2019-05-03:
@djflux Since it is supported upstream it should be fairly straight forward to add it. I've gone ahead and added it to the next sprint.

rusko124, 2019-05-28:
@djflux 
This problem should be solved in the latest development version, try to upgrade. Please let me know if it helped you.

rusko124, 2019-06-12:
I think we can close it.
If there's any issues reopen please.

closed 2019-06-12 by vpetersson