Screenly/Anthias #1100: Viewer crashes when NetworkManager is not in use

joeladria, 2019-05-02:
When filing a bug, please include the following headings if possible. Any example text in this template can be deleted.

#### Overview of the Issue

Just upgraded to the latest production release, and I am not currently a user of NetworManager, however the main loop in viewer.py calls nm-cli even when it was not activated in the install. 

Erasing all code in main() regarding network manager seems to resolve this issue for me. 

#### Reproduction Steps

Steps to reproduce this issue, eg:

* Install OSE using the install script, and opt to NOT use NetworkManager
* viewer does not play content

### Environment

* Raspberry Pi Hardware Version: 

Model 3B+ Revision: 1.3 Ram: 1 GB Sony UK
--

* Screenly OSE Version:production@618864d

ealmonte32, 2019-05-06:
@joeladria 
See my response for fix here:
https://github.com/Screenly/screenly-ose/issues/1102#issuecomment-489695705

rusko124, 2019-06-12:
Should be resolved in the last commits.
I think we can close this

vpetersson, 2019-06-12:
@joeladria Please test the development branch and let us know if the issue still persists.

closed 2019-06-12 by vpetersson