Screenly/Anthias #1102: Viewer Crashes Upon New Update

JeffResc, 2019-05-05:
#### Overview of the Issue

Viewer crashes with many python errors in the output.

#### Reproduction Steps

Simply booted the system running screenly-ose and experienced many python errors.

### Environment

* Raspberry Pi Hardware Version: Raspberry Pi Zero W
* Screenly OSE Version: Commit ff82e476331095b5ad423530ea73732206cbdd94 (Latest at the time of writing)

### Console Output
```
05.05.19 14:38:51 (-0400)  main  May 05 18:38:51 24224b2 python[475]: /usr/local/lib/python2.7/dist-packages/requests/__init__.py:91: RequestsDependencyWarning: urllib3 (1.25.1) or chardet (3.0.4) doesn't match a supported version!
05.05.19 14:38:51 (-0400)  main  May 05 18:38:51 24224b2 python[475]: RequestsDependencyWarning)
05.05.19 14:38:52 (-0400)  main  May 05 18:38:52 24224b2 python[475]: /usr/local/lib/python2.7/dist-packages/cryptography/hazmat/bindings/openssl/binding.py:163: CryptographyDeprecationWarning: OpenSSL version 1.0.1 is no longer supported by the OpenSSL project, please upgrade. A future version of cryptography will drop support for it.
05.05.19 14:38:52 (-0400)  main  May 05 18:38:52 24224b2 python[475]: utils.CryptographyDeprecationWarning
05.05.19 14:38:58 (-0400)  main  May 05 18:38:58 24224b2 python[475]: Viewer crashed.
05.05.19 14:38:58 (-0400)  main  May 05 18:38:58 24224b2 python[475]: Traceback (most recent call last):
05.05.19 14:38:58 (-0400)  main  May 05 18:38:58 24224b2 python[475]: File "/home/pi/screenly/viewer.py", line 516, in <module>
05.05.19 14:38:58 (-0400)  main  May 05 18:38:58 24224b2 python[475]: main()
05.05.19 14:38:58 (-0400)  main  May 05 18:38:58 24224b2 python[475]: File "/home/pi/screenly/viewer.py", line 481, in main
05.05.19 14:38:58 (-0400)  main  May 05 18:38:58 24224b2 python[475]: wireless_connections = nmcli_get_connections('wlan*', 'ScreenlyOSE-*', active=True)
05.05.19 14:38:58 (-0400)  main  May 05 18:38:58 24224b2 python[475]: File "/home/pi/screenly/lib/utils.py", line 123, in nmcli_get_connections
05.05.19 14:38:58 (-0400)  main  May 05 18:38:58 24224b2 python[475]: out = sh.nmcli('-t', '-f', ','.join(fields), 'c', 'show', *params)
05.05.19 14:38:58 (-0400)  main  May 05 18:38:58 24224b2 python[475]: File "/usr/local/lib/python2.7/dist-packages/sh.py", line 1654, in __getattr__
05.05.19 14:38:58 (-0400)  main  May 05 18:38:58 24224b2 python[475]: return self.env[name]
05.05.19 14:38:58 (-0400)  main  May 05 18:38:58 24224b2 python[475]: File "/usr/local/lib/python2.7/dist-packages/sh.py", line 1593, in __getitem__
05.05.19 14:38:58 (-0400)  main  May 05 18:38:58 24224b2 python[475]: return Command._create(k, **self.baked_args)
05.05.19 14:38:58 (-0400)  main  May 05 18:38:58 24224b2 python[475]: File "/usr/local/lib/python2.7/dist-packages/sh.py", line 498, in _create
05.05.19 14:38:58 (-0400)  main  May 05 18:38:58 24224b2 python[475]: if not path: raise CommandNotFound(program)
05.05.19 14:38:58 (-0400)  main  May 05 18:38:58 24224b2 python[475]: CommandNotFound: nmcli
05.05.19 14:38:58 (-0400)  main  May 05 18:38:58 24224b2 python[475]: Traceback (most recent call last):
05.05.19 14:38:58 (-0400)  main  May 05 18:38:58 24224b2 python[475]: File "/home/pi/screenly/viewer.py", line 516, in <module>
05.05.19 14:38:58 (-0400)  main  May 05 18:38:58 24224b2 python[475]: main()
05.05.19 14:38:58 (-0400)  main  May 05 18:38:58 24224b2 python[475]: File "/home/pi/screenly/viewer.py", line 481, in main
05.05.19 14:38:58 (-0400)  main  May 05 18:38:58 24224b2 python[475]: wireless_connections = nmcli_get_connections('wlan*', 'ScreenlyOSE-*', active=True)
05.05.19 14:38:58 (-0400)  main  May 05 18:38:58 24224b2 python[475]: File "/home/pi/screenly/lib/utils.py", line 123, in nmcli_get_connections
05.05.19 14:38:58 (-0400)  main  May 05 18:38:58 24224b2 python[475]: out = sh.nmcli('-t', '-f', ','.join(fields), 'c', 'show', *params)
05.05.19 14:38:58 (-0400)  main  May 05 18:38:58 24224b2 python[475]: File "/usr/local/lib/python2.7/dist-packages/sh.py", line 1654, in __getattr__
05.05.19 14:38:58 (-0400)  main  May 05 18:38:58 24224b2 python[475]: return self.env[name]
05.05.19 14:38:58 (-0400)  main  May 05 18:38:58 24224b2 python[475]: File "/usr/local/lib/python2.7/dist-packages/sh.py", line 1593, in __getitem__
05.05.19 14:38:58 (-0400)  main  May 05 18:38:58 24224b2 python[475]: return Command._create(k, **self.baked_args)
05.05.19 14:38:58 (-0400)  main  May 05 18:38:58 24224b2 python[475]: File "/usr/local/lib/python2.7/dist-packages/sh.py", line 498, in _create
05.05.19 14:38:58 (-0400)  main  May 05 18:38:58 24224b2 python[475]: if not path: raise CommandNotFound(program)
05.05.19 14:38:58 (-0400)  main  May 05 18:38:58 24224b2 python[475]: sh.CommandNotFound: nmcli
05.05.19 14:38:59 (-0400)  main  May 05 18:38:59 24224b2 systemd[1]: screenly-viewer.service: main process exited, code=exited, status=1/FAILURE
05.05.19 14:38:59 (-0400)  main  May 05 18:38:59 24224b2 systemd[1]: Unit screenly-viewer.service entered failed state.
05.05.19 14:38:59 (-0400)  main  May 05 18:38:59 24224b2 systemd[1]: screenly-viewer.service holdoff time over, scheduling restart.
05.05.19 14:38:59 (-0400)  main  May 05 18:38:59 24224b2 systemd[1]: Stopping Screenly Viewer...
05.05.19 14:38:59 (-0400)  main  May 05 18:38:59 24224b2 systemd[1]: Starting Screenly Viewer...
05.05.19 14:38:59 (-0400)  main  May 05 18:38:59 24224b2 systemd[1]: Started Screenly Viewer.
```

vpetersson, 2019-05-06:
@JeffResc what installation method did you use? Also, did you upgrade after the installation?

JeffResc, 2019-05-06:
I deployed on Balena following [their documentation](https://www.balena.io/blog/deploy-a-digital-signage-application-with-screenly-and-resin/)

vpetersson, 2019-05-06:
@JeffResc Those instructions are unfortunately fairly out of date. We've dedicated [the next sprint](https://github.com/Screenly/screenly-ose/projects/3) to resolve those issues. In short, Balena moved to a multi-container setup, which broke a lot of things for us.

JeffResc, 2019-05-06:
I didn’t see any other Balena options in the README. Is it still possible to run screenly-ose on Balena devices? If so, should the proper steps be included in the README?

vpetersson, 2019-05-06:
That's what the next sprint is all about. In short, right now the Balena support is broken.

JeffResc, 2019-05-06:
Ah, you do know which commit I can rollback to that I can use for Balena support in the meantime?

rusko124, 2019-05-06:
@JeffResc Hi!
You can look the progress here #1099 
I hope it will be in the master soon.

JeffResc, 2019-05-08:
@ealmonte32 I tried adding the repository using the `add-apt-repository` command but received this: `/bin/sh: 1: add-apt-repository: not found`. Am I doing something wrong?

ealmonte32, 2019-05-08:
> @ealmonte32 I tried adding the repository using the `add-apt-repository` command but received this: `/bin/sh: 1: add-apt-repository: not found`. Am I doing something wrong?

Yes, you need to edit your apt sources file and add that URL.
`/etc/apt/sources.list`

JeffResc, 2019-05-08:
Hm. Everything with the apt repo worked out fine, but receiving some errors from pip still:
```
[main]     ERROR: requests 2.20.0 has requirement urllib3<1.25,>=1.21.1, but you'll have urllib3 1.25.1 which is incompatible.
[main]     ERROR: cryptography 2.6.1 has requirement cffi!=1.11.3,>=1.8, but you'll have cffi 1.7.0 which is incompatible.
```

JeffResc, 2019-05-09:
Just built with those pip errors and it seems like I'm still encountering the same errors:
```
08.05.19 20:58:46 (-0400)  main  May 09 00:58:46 38dd725 python[2716]: /usr/local/lib/python2.7/dist-packages/requests/__init__.py:91: RequestsDependencyWarning: urllib3 (1.25.1) or chardet (3.0.4) doesn't match a supported version!
08.05.19 20:58:46 (-0400)  main  May 09 00:58:46 38dd725 python[2716]: RequestsDependencyWarning)
08.05.19 20:58:47 (-0400)  main  May 09 00:58:47 38dd725 python[2716]: /usr/local/lib/python2.7/dist-packages/cryptography/hazmat/bindings/openssl/binding.py:163: CryptographyDeprecationWarning: OpenSSL version 1.0.1 is no longer supported by the OpenSSL project, please upgrade. A future version of cryptography will drop support for it.
08.05.19 20:58:47 (-0400)  main  May 09 00:58:47 38dd725 python[2716]: utils.CryptographyDeprecationWarning
08.05.19 20:58:53 (-0400)  main  May 09 00:58:53 38dd725 python[2716]: Viewer crashed.
08.05.19 20:58:53 (-0400)  main  May 09 00:58:53 38dd725 python[2716]: Traceback (most recent call last):
08.05.19 20:58:53 (-0400)  main  May 09 00:58:53 38dd725 python[2716]: File "/home/pi/screenly/viewer.py", line 516, in <module>
08.05.19 20:58:53 (-0400)  main  May 09 00:58:53 38dd725 python[2716]: main()
08.05.19 20:58:53 (-0400)  main  May 09 00:58:53 38dd725 python[2716]: File "/home/pi/screenly/viewer.py", line 481, in main
08.05.19 20:58:53 (-0400)  main  May 09 00:58:53 38dd725 python[2716]: wireless_connections = nmcli_get_connections('wlan*', 'ScreenlyOSE-*', active=True)
08.05.19 20:58:53 (-0400)  main  May 09 00:58:53 38dd725 python[2716]: File "/home/pi/screenly/lib/utils.py", line 123, in nmcli_get_connections
08.05.19 20:58:53 (-0400)  main  May 09 00:58:53 38dd725 python[2716]: out = sh.nmcli('-t', '-f', ','.join(fields), 'c', 'show', *params)
08.05.19 20:58:53 (-0400)  main  May 09 00:58:53 38dd725 python[2716]: File "/usr/local/lib/python2.7/dist-packages/sh.py", line 1654, in __getattr__
08.05.19 20:58:53 (-0400)  main  May 09 00:58:53 38dd725 python[2716]: return self.env[name]
08.05.19 20:58:53 (-0400)  main  May 09 00:58:53 38dd725 python[2716]: File "/usr/local/lib/python2.7/dist-packages/sh.py", line 1593, in __getitem__
08.05.19 20:58:53 (-0400)  main  May 09 00:58:53 38dd725 python[2716]: return Command._create(k, **self.baked_args)
08.05.19 20:58:53 (-0400)  main  May 09 00:58:53 38dd725 python[2716]: File "/usr/local/lib/python2.7/dist-packages/sh.py", line 498, in _create
08.05.19 20:58:53 (-0400)  main  May 09 00:58:53 38dd725 python[2716]: if not path: raise CommandNotFound(program)
08.05.19 20:58:53 (-0400)  main  May 09 00:58:53 38dd725 python[2716]: CommandNotFound: nmcli
08.05.19 20:58:53 (-0400)  main  May 09 00:58:53 38dd725 python[2716]: Traceback (most recent call last):
08.05.19 20:58:53 (-0400)  main  May 09 00:58:53 38dd725 python[2716]: File "/home/pi/screenly/viewer.py", line 516, in <module>
08.05.19 20:58:53 (-0400)  main  May 09 00:58:53 38dd725 python[2716]: main()
08.05.19 20:58:53 (-0400)  main  May 09 00:58:53 38dd725 python[2716]: File "/home/pi/screenly/viewer.py", line 481, in main
08.05.19 20:58:53 (-0400)  main  May 09 00:58:53 38dd725 python[2716]: wireless_connections = nmcli_get_connections('wlan*', 'ScreenlyOSE-*', active=True)
08.05.19 20:58:53 (-0400)  main  May 09 00:58:53 38dd725 python[2716]: File "/home/pi/screenly/lib/utils.py", line 123, in nmcli_get_connections
08.05.19 20:58:53 (-0400)  main  May 09 00:58:53 38dd725 python[2716]: out = sh.nmcli('-t', '-f', ','.join(fields), 'c', 'show', *params)
08.05.19 20:58:53 (-0400)  main  May 09 00:58:53 38dd725 python[2716]: File "/usr/local/lib/python2.7/dist-packages/sh.py", line 1654, in __getattr__
08.05.19 20:58:53 (-0400)  main  May 09 00:58:53 38dd725 python[2716]: return self.env[name]
08.05.19 20:58:53 (-0400)  main  May 09 00:58:53 38dd725 python[2716]: File "/usr/local/lib/python2.7/dist-packages/sh.py", line 1593, in __getitem__
08.05.19 20:58:53 (-0400)  main  May 09 00:58:53 38dd725 python[2716]: return Command._create(k, **self.baked_args)
08.05.19 20:58:53 (-0400)  main  May 09 00:58:53 38dd725 python[2716]: File "/usr/local/lib/python2.7/dist-packages/sh.py", line 498, in _create
08.05.19 20:58:53 (-0400)  main  May 09 00:58:53 38dd725 python[2716]: if not path: raise CommandNotFound(program)
08.05.19 20:58:53 (-0400)  main  May 09 00:58:53 38dd725 python[2716]: sh.CommandNotFound: nmcli
08.05.19 20:58:54 (-0400)  main  May 09 00:58:54 38dd725 systemd[1]: screenly-viewer.service: main process exited, code=exited, status=1/FAILURE
08.05.19 20:58:54 (-0400)  main  May 09 00:58:54 38dd725 systemd[1]: Unit screenly-viewer.service entered failed state.
08.05.19 20:58:54 (-0400)  main  May 09 00:58:54 38dd725 systemd[1]: screenly-viewer.service holdoff time over, scheduling restart.
08.05.19 20:58:54 (-0400)  main  May 09 00:58:54 38dd725 systemd[1]: Stopping Screenly Viewer...
08.05.19 20:58:54 (-0400)  main  May 09 00:58:54 38dd725 systemd[1]: Starting Screenly Viewer...
08.05.19 20:58:55 (-0400)  main  May 09 00:58:55 38dd725 systemd[1]: Started Screenly Viewer.
```

ealmonte32, 2019-05-09:
@JeffResc 
You are running into two separate issues here.

1. nmcli error means you dont have network-manager installed, even if you dont use it, the new screenly was coded to require network-manager, @rusko124  is working on making it not a requirement though, so to easily fix that, just do `sudo apt install network-manager`

2. the dependency errors you are getting is because of what I mentioned before, that pip install and apt install conflict each other when they install packages, to see what I mean type this:
`dpkg -l | grep 'chardet\|urllib' && pip list | grep 'chardet\|urllib'`

As you should see, you will have conflicting versions, most likely the dpkg list will show you the old versions of the required packages..

JeffResc, 2019-05-09:
@ealmonte32 
Although I've appreciated all the help you've given me, I've installed @rusko124's pull request onto my devices and they're now working fine. Thank you for everything!

ealmonte32, 2019-05-09:
> @ealmonte32
> Although I've appreciated all the help you've given me, I've installed @rusko124's pull request onto my devices and they're now working fine. Thank you for everything!

Thats fine, I was just trying to show you one way to fix it manually, so that if you ever encounter package dependencies again, you would know a way to fix it without relying on reinstalling the whole system.

rusko124, 2019-06-12:
Can we close this?

closed 2019-06-12 by vpetersson