Screenly/Anthias #1103: I can't find the controlpanel

TheCazz, 2019-05-06:
Hi
I have download the latest image of screenly OSE but I can't see where I can reboot or shutdown the raspberry.

Have look in the Settings and System Info

thefyfy, 2019-05-06:
Hi
Sorry but this is not possible. The Screenly OSE API doesn't expose this services.
(and this is not available in Settings and System Info pages...)

vpetersson, 2019-05-06:
There is no functionality for this exposed. You'd need to SSH into the device and shut it down.

TheCazz, 2019-05-06:
Hmmm then I misunderstood when I read this which should be included in the latest version
https://github.com/Screenly/screenly-ose/issues/121

rusko124, 2019-05-06:
@TheCazz Hi!
Currently it works in the celery branch. In this sprint, we plan to fully test it then it will be in the production.

TheCazz, 2019-05-06:
Ahh thanks :)

closed 2019-05-16 by vpetersson