Screenly/Anthias #1104: Screenly completely stop showing assets

qbicdesign, 2019-05-07:
I have a mixture of asset types, URLs and Videos. I noticed that my TV screen has been black for a few days. When I finally got around to looking at it, I accessed the web UI and checked the System info.  Last entry in the log is 29th April , yet all assets are not set to expire until 2024

#### Reproduction Steps

Steps to reproduce this issue, eg:

* Add a bunch of assets
* Wait a few days

### Environment

* Raspberry Pi Hardware Version: Model 3B+ Revision: 1.3 Ram: 1 GB Sony UK
* Screenly OSE Version: development@79b0818

ealmonte32, 2019-05-13:
Some logs would help troubleshoot this.. are the videos uploaded as files or as a URL? asking because remember ram on Pi is low.. use command `htop` to see how much memory is being used at least once at end of day.. if this is an issue, I recommend using ZRAM https://github.com/novaspirit/rpi_zram

qbicdesign, 2019-05-21:
I've upgraded to experimental now, and the issue is gone. Closing this

closed 2019-05-21 by qbicdesign