Screenly/Anthias #1108: Stretching size of website .gif file

dkopyone, 2019-05-08:
Raspberry Pi 3 Model B
Screenly OSE version Sprint 9

I have a website that has an animated gif weather map that gets updated every hour.  I need to display this with screenly and reload the page every hour to get the updated image.  It works fine except the image is really small, about half the size of the screen.  Is there any way to have it automatically stretch the gif to fill the screen?  I have no way of editing the size of the gif myself, since its on a weather website.

vpetersson, 2019-05-09:
I'd suggest you try another weather widget instead. For instance, check out our [weather widget](https://app-store.srly.io/weather/) that is designed for signage usage.

Generally speaking, regular websites are not good to use for signage. You need content that is optimized for this particular use case.

madsrh, 2019-05-16:
> ...Is there any way to have it automatically stretch the gif to fill the screen?...

@dkopyone you can just create your own website that grabs the .gif like this:

https://gist.github.com/leemark/5204533

> ...You need **content that is optimized** for this particular use case.

Since there's no issue with Screenly, can we close this issue?

vpetersson, 2019-05-16:
@madsrh Agreed. Closing.

closed 2019-05-16 by vpetersson