Screenly/Anthias #1109: Way to manage assets in RAM?

JeffResc, 2019-05-09:
When using the same asset in loop, I noticed that the black screen will still display between loops. Since my entire playlist is just that one asset, is it possible to keep that asset loaded in RAM to prevent this black screen inbetween the loop?

vpetersson, 2019-05-09:
@JeffResc this is unforrtunately how `omxplayer` works. There is a loop feature that could in theory be used if you only have one asset, but we have not adopted it as it is somewhat of an edge case. Moving things to RAM may possibly improve it, but not as much as the loop feature.

JeffResc, 2019-05-09:
@vpetersson Are you aware of what file(s) I could modify in order to gain that functionality on my own personal build?

vpetersson, 2019-05-09:
@JeffResc you'd have to modify the `view_video()` function in `viewer.py` and add logic along the lines of "if the playlist only includes one asset and that one asset is a video, then append the flag `--loop`"

That said, it is possible that this will break other fuctionalities, such as if you add another asset, so it needs to be tested fairly extensibly for edge cases.

JeffResc, 2019-05-09:
Hm, I understand, but I also don’t. You stated the if statement that would need to be implemented, but then stated that it could break other functionality with adding other assets. With that if statement in place, it should work okay?

vpetersson, 2019-05-09:
Well, the problem is the if the `--loop` flag is appended, the player will get stuck in an infinite loop. Hence we would need a way to break out of this if the playlist is modified. There are some checks in place for this, but I don't know if thay would solve this particular situation.

JeffResc, 2019-05-09:
@vpetersson Perhaps just ending the process or is there a way to request it to stop?

vpetersson, 2019-05-10:
It's been a while since I touched this code, but I believe the appropriate solution is to send perhaps a SIGKILL or SIGHUP (don't remember how signals are implemented) to omxplayer when the playlist change, which should then kick it out of the infinite loop.

ealmonte32, 2019-05-13:
@JeffResc would stopping the process with the simple `sudo systemctl stop screenly-viewer.service` be enough of a workaround for you whenever you need to kill the process?

vpetersson, 2019-05-13:
@ealmonte32 I don't think that's a good way to do it. We already have some signalling support built into the viewer, so it should b efairly straight forward to implement it properly (I think).

ealmonte32, 2019-05-13:
@vpetersson Hmm.. but doesnt SIGKILL require process ID? since to kill by process name you need pkill instead.. just wondering what is easier..

vpetersson, 2019-05-13:
That likely depends on the library. However, in theory we could invoke `pkill`, it would just be less clean than sending a signal using Python directly.

ealmonte32, 2019-05-13:
Ah, you want him to use viewer.py instead of something separate, gotcha.

closed 2019-07-19 by stale[bot]