Screenly/Anthias #1111: Blank Screen

rhodesmi, 2019-05-11:
I updated our Screenly OSE this afternoon and now we have a blank screen. Other than that its fine! 

more foobar.txt foobar.txt [Kfoobar.txt [Kfoobar.txt [Kfoobar.txt [Kfoobar.txt [Kcfoobar.txt afoobar.txt tfoobar.txt  foobar.txt 
-- Logs begin at Sat 2019-05-11 16:02:15 UTC, end at Sat 2019-05-11 16:03:17 UTC. --
May 11 16:02:15 raspberrypi kernel: Booting Linux on physical CPU 0x0
May 11 16:02:15 raspberrypi kernel: Linux version 4.14.98+ (dom@dom-XPS-13-9370) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1200 Tue Feb 12 20:11:02 GMT 2019
May 11 16:02:15 raspberrypi kernel: CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
May 11 16:02:15 raspberrypi kernel: CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
May 11 16:02:15 raspberrypi kernel: OF: fdt: Machine model: Raspberry Pi Model B Rev 2
May 11 16:02:15 raspberrypi kernel: Memory policy: Data cache writeback
May 11 16:02:15 raspberrypi kernel: cma: Reserved 8 MiB at 0x17400000
May 11 16:02:15 raspberrypi kernel: On node 0 totalpages: 98304
May 11 16:02:15 raspberrypi kernel: free_area_init_node: node 0, pgdat c09caaf0, node_mem_map d7c8a200
May 11 16:02:15 raspberrypi kernel:   Normal zone: 864 pages used for memmap
May 11 16:02:15 raspberrypi kernel:   Normal zone: 0 pages reserved
May 11 16:02:15 raspberrypi kernel:   Normal zone: 98304 pages, LIFO batch:31
May 11 16:02:15 raspberrypi kernel: pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
May 11 16:02:15 raspberrypi kernel: pcpu-alloc: [0] 0 
May 11 16:02:15 raspberrypi kernel: Built 1 zonelists, mobility grouping on.  Total pages: 97440
May 11 16:02:15 raspberrypi kernel: Kernel command line: bcm2708_fb.fbwidth=640 bcm2708_fb.fbheight=480 bcm2708_fb.fbdepth=32 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=PARTUUID=597532ff-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait splash vt.global_cursor_default=0 quiet init=/lib/systemd/systemd net.ifnames=0
May 11 16:02:15 raspberrypi kernel: PID hash table entries: 2048 (order: 1, 8192 bytes)
May 11 16:02:15 raspberrypi kernel: Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
May 11 16:02:15 raspberrypi kernel: Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
May 11 16:02:15 raspberrypi kernel: Memory: 370248K/393216K available (6451K kernel code, 589K rwdata, 1992K rodata, 440K init, 673K bss, 14776K reserved, 8192K cma-reserved)
May 11 16:02:15 raspberrypi kernel: Virtual kernel memory layout:
                                        vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                                        fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                                        vmalloc : 0xd8800000 - 0xff800000   ( 624 MB)
                                        lowmem  : 0xc0000000 - 0xd8000000   ( 384 MB)
                                        modules : 0xbf000000 - 0xc0000000   (  16 MB)
                                          .text : 0xc0008000 - 0xc0655050   (6453 kB)
                                          .init : 0xc08d4000 - 0xc0942000   ( 440 kB)
                                          .data : 0xc0942000 - 0xc09d54f8   ( 590 kB)
                                           .bss : 0xc09dadf0 - 0xc0a8338c   ( 674 kB)
May 11 16:02:15 raspberrypi kernel: SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
May 11 16:02:15 raspberrypi kernel: ftrace: allocating 24128 entries in 71 pages
May 11 16:02:15 raspberrypi kernel: NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
May 11 16:02:15 raspberrypi kernel: sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
May 11 16:02:15 raspberrypi kernel: clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
May 11 16:02:15 raspberrypi kernel: bcm2835: system timer (irq = 27)
May 11 16:02:15 raspberrypi kernel: Console: colour dummy device 80x30
May 11 16:02:15 raspberrypi kernel: console [tty1] enabled
May 11 16:02:15 raspberrypi kernel: Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
May 11 16:02:15 raspberrypi kernel: pid_max: default: 32768 minimum: 301
May 11 16:02:15 raspberrypi kernel: Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
May 11 16:02:15 raspberrypi kernel: Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
May 11 16:02:15 raspberrypi kernel: Disabling memory control group subsystem
May 11 16:02:15 raspberrypi kernel: CPU: Testing write buffer coherency: ok
May 11 16:02:15 raspberrypi kernel: Setting up static identity map for 0x8200 - 0x8238
May 11 16:02:15 raspberrypi kernel: devtmpfs: initialized
May 11 16:02:15 raspberrypi kernel: random: get_random_u32 called from bucket_table_alloc+0x88/0x1c4 with crng_init=0
May 11 16:02:15 raspberrypi kernel: VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
May 11 16:02:15 raspberrypi kernel: clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
May 11 16:02:15 raspberrypi kernel: futex hash table entries: 256 (order: -1, 3072 bytes)
May 11 16:02:15 raspberrypi kernel: pinctrl core: initialized pinctrl subsystem
May 11 16:02:15 raspberrypi kernel: NET: Registered protocol family 16
May 11 16:02:15 raspberrypi kernel: DMA: preallocated 1024 KiB pool for atomic coherent allocations
May 11 16:02:15 raspberrypi kernel: hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
May 11 16:02:15 raspberrypi kernel: hw-breakpoint: maximum watchpoint size is 4 bytes.
May 11 16:02:15 raspberrypi kernel: Serial: AMBA PL011 UART driver
May 11 16:02:15 raspberrypi kernel: bcm2835-mbox 2000b880.mailbox: mailbox enabled
May 11 16:02:15 raspberrypi kernel: bcm2835-dma 20007000.dma: DMA legacy API manager at d880d000, dmachans=0x1
May 11 16:02:15 raspberrypi kernel: SCSI subsystem initialized
May 11 16:02:15 raspberrypi kernel: usbcore: registered new interface driver usbfs
May 11 16:02:15 raspberrypi kernel: usbcore: registered new interface driver hub
May 11 16:02:15 raspberrypi kernel: usbcore: registered new device driver usb
May 11 16:02:15 raspberrypi kernel: raspberrypi-firmware soc:firmware: Attached to firmware from 2019-03-27 15:45
May 11 16:02:15 raspberrypi kernel: clocksource: Switched to clocksource timer
May 11 16:02:15 raspberrypi kernel: VFS: Disk quotas dquot_6.6.0
May 11 16:02:15 raspberrypi kernel: VFS: Dquot-cache hash table entries: 1024 (order 0, 4096 bytes)
[... 516 lines cut ...]
May 11 16:03:10 raspberrypi sshd[434]: Accepted password for pi from 192.168.1.174 port 63084 ssh2
May 11 16:03:10 raspberrypi sshd[434]: pam_unix(sshd:session): session opened for user pi by (uid=0)
May 11 16:03:10 raspberrypi systemd[1]: Created slice User Slice of pi.
May 11 16:03:10 raspberrypi systemd[1]: Starting User Manager for UID 1000...
May 11 16:03:10 raspberrypi systemd[443]: pam_unix(systemd-user:session): session opened for user pi by (uid=0)
May 11 16:03:10 raspberrypi systemd[1]: Started Session c1 of user pi.
May 11 16:03:10 raspberrypi systemd-logind[177]: New session c1 of user pi.
May 11 16:03:11 raspberrypi systemd[443]: Listening on GnuPG cryptographic agent (access for web browsers).
May 11 16:03:11 raspberrypi systemd[443]: Reached target Paths.
May 11 16:03:11 raspberrypi systemd[443]: Listening on GnuPG cryptographic agent and passphrase cache.
May 11 16:03:11 raspberrypi systemd[443]: Listening on GnuPG cryptographic agent and passphrase cache (restricted).
May 11 16:03:11 raspberrypi systemd[443]: Reached target Timers.
May 11 16:03:11 raspberrypi systemd[443]: Listening on GnuPG cryptographic agent (ssh-agent emulation).
May 11 16:03:11 raspberrypi systemd[443]: Listening on GnuPG network certificate management daemon.
May 11 16:03:11 raspberrypi systemd[443]: Reached target Sockets.
May 11 16:03:11 raspberrypi systemd[443]: Reached target Basic System.
May 11 16:03:11 raspberrypi systemd[443]: Reached target Default.
May 11 16:03:11 raspberrypi systemd[443]: Startup finished in 409ms.
May 11 16:03:11 raspberrypi systemd[1]: Started User Manager for UID 1000.
May 11 16:03:12 raspberrypi sshd[434]: lastlog_openseek: Couldn't stat /var/log/lastlog: No such file or directory
May 11 16:03:12 raspberrypi sshd[434]: lastlog_openseek: Couldn't stat /var/log/lastlog: No such file or directory
May 11 16:03:17 raspberrypi sudo[474]:       pi : TTY=pts/0 ; PWD=/home/pi ; USER=root ; COMMAND=/bin/journalctl
May 11 16:03:17 raspberrypi sudo[474]: pam_unix(sudo:session): session opened for user root by pi(uid=0)
pi@raspberrypi:~$ 
[pi.txt](https://github.com/Screenly/screenly-ose/files/3169245/pi.txt)

ealmonte32, 2019-05-13:
Wow, lots of errors.. are you sure your Pi was able to properly update and finish everything? I've never seen so many failed messages before after running the screenly install script..

rhodesmi, 2019-05-13:
Todays lesson - It's quicker to put a new Image on than do an onbox upgrade (only took 5 mins to install a new image versus the two hours to upgrade it.) So now with a new image it's all working fine,

closed 2019-05-13 by rhodesmi

vpetersson, 2019-05-13:
@rhodesmi This is not all that surprising unfortunately (and why we provide disk images). The Raspberry Pi isn't a very powerful device, so upgrades can take a long time (in particular for older Raspberry Pis).