Screenly/Anthias #1114: picture in picture view of actual screen

pooh22, 2019-05-13:
#### Overview of the Issue

When working from home or another location without direct view of the screen, a "picture in picture" showing what is currently actually on the screen would be of great help.

#### How it would work

In the schedule overview, a small area shows what is currently shown on the screen driven by screenly. It would be even better if the display status (TV on/standby/powered-off) would be shown as well.

vpetersson, 2019-05-13:
Take a look at #1030 for some inspiration.

(Side-note: this is already included in the commercial version of Screenly)

pooh22, 2019-05-14:
Thanks for the pointer, I guess the VNC option would be temporary anyway, because I read it would be implemented differently in the upcoming version. And VNC is probably not feasible considering the networking requirements...

Would an approach using the API work? I noticed there's an API interface to see which asset is active, however I don't understand how the API works yet. Can the API be used to get the asset data itself? It might then be possible to make a webpage with autorefresh getting the active asset and displaying it in a browser (or an iframe or whatever)...
(slightly out of my comfort zone here ;-) I'm not a website developer, just a sysadmin ;-)

vpetersson, 2019-05-14:
@pooh22 In theory, i guess you could check for the currently active asset over the API and then pull down said asset. However, it's not an optimal approach from a bandwidth perspective. 

VNC is an interesting solution, but it comes with its own set of challenges.

closed 2019-07-20 by stale[bot]