Screenly/Anthias #1115: After move /var/log/ to ramdisk management page is not reachable

urb5tar, 2019-05-13:
#### Overview of the Issue

When I try to move the directory /var/log to ramdisk then the screenly management page is not reachable

#### Reproduction Steps
edit /etc/fstab 

insert line 
none        /var/log        tmpfs   size=5M,noatime         00

reboot

### Environment

* Raspberry Pi Hardware Version: Raspberry Pi 3 B+
* Screenly OSE Version: 2019-04-27-Screenly-OSE Image

vpetersson, 2019-05-13:
That's strange. @rusko124 are you able to reproduce this?

rusko124, 2019-06-12:
Yes, i can reproduce it.
Need to setup path to nginx log files
```
nginx: [alert] could not open error log file: open() "/var/log/nginx/error.log" failed (2: No such file or directory)
2019/06/12 06:17:13 [emerg] 1548#1548: open() "/var/log/nginx/access.log" failed (2: No such file or directory)
```

vpetersson, 2019-06-12:
@urb5tar actually, looking closer at this, this isn't really not unexpected. Just check the logs and create the required directories. I don't think this should be part of Screenly OSE's code base.

closed 2019-06-12 by vpetersson

urb5tar, 2019-06-12:
Thank you very much for figuring this out