Screenly/Anthias #1116: Web Interface Accessibility

wcbviwebmaster, 2019-05-14:
The following basic accessibility improvements are needed in the web interface.

1.  The Screenly logo needs image alt text for screen reader users

2.  The asset download, edit, and delete buttons need to be labeled.  My screen reader currently just says "button" for these items.

3.  The blue color used for links, the top bar, active and inactive asset headers, and on/off buttons does not meet WCAG level AA color contrast guidelines.  The blue color could be changed slightly to #0072ED and it would then meet level AA color contrast guidelines.

4.  The previous asset button is a heading.  This seems odd to me and not a good use of headers, but I supposed that is my opinion.  I would think that the active assets and inactive assets headers should also be coded as a heading as well.

I'm new to using GitHub as this is my first issue.  I do know HTML & CSS and am willing to help make these changes if desired.
Overall the interface is mostly accessible other than these issues.

I'm using Screenly OSE v0.17 on a Raspberry Pi 3 Model B

vpetersson, 2019-05-14:
@wcbviwebmaster Good catch! Yes, we haven't really tested things for screen readers, so we'd be more than happy to accept a pull request that addresses these issues.

clbuggs, 2019-05-14:
@vpetersson thank you.  I'm learning more about pull requests in preparation for contributing.

The other accessibility issue I forgot to mention deals with the asset play order.  Currently, it is necessary to "drag-and-drop" assets to arrange the order that they are played.  This "drag-and-drop" functionality is not available to keyboard-only or screen reader users.  The way to fix this would be to implement a "weight" column to specify a value (1, 2, 3, etc.)  Some content management systems like Drupal have a button to switch between the drag-and-drop and item weight behavior to make everyone happy.

closed 2019-07-20 by stale[bot]