Screenly/Anthias #1117: Fix "Screenly Version" for Balena deployments

vpetersson, 2019-05-15:
Currently, the "Screenly Version" field under System Info is broken on Balena. Unfortunately, there doesn't seem to be a way to [read the version](https://www.balena.io/docs/learn/develop/runtime/) using environment variables, so I guess we need to write the version to disk and read it back.

ShukhratK, 2019-05-15:
Hi. Is this version compatible with Balena? According to this https://www.balena.io/blog/deploy-a-digital-signage-application-with-screenly-and-resin/ it is not.

vpetersson, 2019-05-15:
@ShukhratK Well, technically speaking, it was Balena/Resin that broke the support. It has since been fixed again.

ShukhratK, 2019-05-15:
@vpetersson thx for ur prompt response. i just tried it and got the following error:

`[main]     W: Failed to fetch http://deb.debian.org/debian/dists/jessie-updates/InRelease  Unable to find expected entry 'main/binary-armhf/Packages' in Release file (Wrong sources.list entry or malformed file)
[main]     E: Some index files failed to download. They have been ignored, or old ones used instead.
[main]
[main]     Removing intermediate container abb766e57ee2
[Info]     Uploading images
[main]     The command '/bin/sh -c apt-get update &&     apt-get -y install         build-essential         curl         git-core         libffi-dev         libssl-dev         matchbox         net-tools         nginx-light         omxplayer         psmisc         python-dev         python-imaging         python-netifaces         python-simplejson         libraspberrypi0         ifupdown         sqlite3         uzbl         x11-xserver-utils         xserver-xorg &&     apt-get clean' returned a non-zero code: 100
[Success]  Successfully uploaded images
[Error]    Some services failed to build:
[Error]      Service: main
[Error]        Error: The command '/bin/sh -c apt-get update &&     apt-get -y install         build-essential         curl         git-core         libffi-dev         libssl-dev         matchbox         net-tools         nginx-light         omxplayer         psmisc         python-dev         python-imaging         python-netifaces         python-simplejson         libraspberrypi0         ifupdown         sqlite3         uzbl         x11-xserver-utils         xserver-xorg &&     apt-get clean' returned a non-zero code: 100
[Error]    Not deploying release.

remote: error: hook declined to update refs/heads/master
To git.balena-cloud.com:gh_shukhratk/PiTest3.git
! [remote rejected] master -> master (hook declined)
error: failed to push some refs to 'gh_shukhratk@git.balena-cloud.com:gh_shukhratk/PiTest3.git'`

vpetersson, 2019-05-15:
@ShukhratK From the logs it looks like you're on the production release. This has not yet been merged into there, so switch to the master branch, pull down the latest changes, and you should be good to go.

ShukhratK, 2019-05-15:
@vpetersson 
sounds good
thx again

ShukhratK, 2019-05-15:
@vpetersson same error with the master fork

vpetersson, 2019-05-15:
Can you paste the output of the following commands?

```
$ git pull
$ git rev-parse --short HEAD 
$ git branch
$ git status
```

ShukhratK, 2019-05-15:
Already up to date.
79b0818
* master
On branch master
Your branch is up to date with 'origin/master'.
nothing to commit, working tree clean

vpetersson, 2019-05-15:
Are you using a fork? If so, that would explain it. The latest upstream hash on master is 093f594. Your hash, (79b0818) does not have the Balena fixes in it.

ShukhratK, 2019-05-15:
i see. how do i fix that?

ShukhratK, 2019-05-15:
@vpetersson i am using a fork.

vpetersson, 2019-05-15:
Yeah so you either need to [merge in the latest changes](https://help.github.com/en/articles/syncing-a-fork) or not use a fork.

ShukhratK, 2019-05-15:
sorry, this is probably a noobie question: i am getting the following error now

fatal: 'upstream' does not appear to be a git repository

vpetersson, 2019-05-15:
Please move this discussion to the [forum](https://forums.screenly.io/) as this is out of scope for this particular issue.

ShukhratK, 2019-05-15:
will do. thx again

vpetersson, 2019-05-16:
Closing in favour of  #1126/

closed 2019-05-16 by vpetersson